keyboardio/Chrysalis #376: Make modifier checkboxes available to all key types

leavesofgrass, 2019-04-24:
You can click a checkbox on a digit or a letter to add a modifier like Alt, Shift, or CTRL but this functionality isn't available to Miscellaneous keys like PrnScr or the Mouse button keys.

[image]

Including these modifiers for all key types would be ideal in order to maximize mapping potential. 

(CTRL+SHIFT+PrnScr is also "copy a screenshot of the selected area to the clipboard" in Ubuntu.)

_On a related note_: Please consider treating Insert as such a modifier. It would be a huge plus for many people with disabilities because it is used so extensively with accessibility software as a critical modifier.

Exitialis1, 2019-06-22:
I'd also like this, Alt + print screen is take a screenshot of the selected window in Windows and I prefer having that setup as a single key press.

hnrklssn, 2019-07-01:
Seems closely related to this issue: https://github.com/keyboardio/Chrysalis/issues/257
Mostly commenting just to link them together.

tmarzolf, 2020-06-18:
Alt + print screen would be specifically useful to me as well as very desirable as a single key press in Windows...

obra, 2020-06-23:
I think the only limitation is that modifiers should not be available on consumercontrol/systemcontrol keys

bjartek, 2020-10-08:
Yes please :)

algernon, 2021-01-25:
This is now fixed on master, and in 0.8.1-snapshot.

closed 2021-01-25 by algernon