keyboardio/Chrysalis #377: Demonstration mode

Ptepp1c, 2019-04-27:
**Is your feature request related to a problem? Please describe.**
I am interested in purchasing the Keyboardio Model 01, but cannot seem to test the Chrysalis software as I do not have a compatible keyboard

**Describe the solution you'd like**
It would be nice to have a demonstration mode, so that you can try out the program when you don't have a compatible keyboard on hand, it would be great if this demo mode has the option to have a range of example keyboards based on those that can be used with chrysallis

**Describe alternatives you've considered**
I have read around on forums about Crysalis, downloaded the .exe and tried to see if there was any demo mode, I have poked around a bit on github, looking at the readme etc. (Note I have not tried to launch a development environment because it is beyond me at this point in time, I thought Yarn was to do with knitting :) )

**Additional context**
This is a very low priority feature request I am sure you already have such a feature in mind when the program is made live, once you have finished alpha and beta.

TreTuna, 2019-04-28:
Not quite a "demo" mode, but making Chrysalis usable without a keyboard connected is actually decently high on my own list. I don't always have my Model01 with me when I want to work on Chrysalis, so want to get it set to where I can code on it without that.

TreTuna, 2019-04-28:
We've definitely got some steps ahead before we get there, and algernon and myself haven't been able to dedicate much time to this project lately, but hopefully soon again.

mattvenn, 2019-04-30:
would also be very useful for allowing contributors to work without having access to the keyboard

0957758592, 2019-08-12:
under development

Here is a preview: 
https://www.loom.com/share/3a4d4e0d30e943b3808c114224c4ec01

obra, 2019-08-19:
I'm happy this is being worked on. I know @tretuna has some strong feelings aobut the right way to implement this.

In terms of UI, rather than a "Demo mode" toggle on the front page, perhaps we could always add them to the dropdown under a "Demo Mode" section?

TreTuna, 2019-08-22:
Ultimately, there shouldn't be a "demo" mode, but Chrysalis should be able to work offline from a keyboard (only connecting when we need to read/sync/write). That's the direction I hope to take this when I get a chance, it's not an easy or simple fix. Sadly I'm under a big deadline with my work atm for the end of Sept so I don't think I have much time until then to pursue this objective. I think @0957758592's solution would work for now until we could rework the internals to work more disconnected from the board.

alexryndin, 2021-06-27:
Any progress on this?

algernon, 2022-06-26:
We've made _some_ progress towards this feature. To be able to implement this in a way that is maintainable, we need to rework a lot of the Chrysalis internals. We've been slowly refactoring things for a while now, but we're still quite far from being able to start working on an "offline" mode like proposed here.

It is on our radar, and it is an important feature to have, but it's not a top priority yet. I want to get [Chrysalis 1.0](https://github.com/keyboardio/Chrysalis/milestone/8) out the door first. Once that's out, we'll build a roadmap for the next version, and I suspect this'll be pretty high on the list then.