keyboardio/Chrysalis #378: allow different keyboard layouts in one hardware type

mattvenn, 2019-04-30:
The Dygma Raise has ISO and ANSI versions. Hardware type can be queried via serial.

The configurator should automatically choose which key layout to display to the user.

TreTuna, 2019-05-04:
This could go in the right corner in replacement of the keyboard name tag that's talked about removing in https://github.com/keyboardio/Chrysalis/issues/385 or in the spot where the layers dropdown currently is if we move that over to join the other layer buttons.

mattvenn, 2019-05-14:
It shouildn't need a control, because the configurator can request the type of hardware from the keyboard.

mattvenn, 2019-05-30:
OK, we now have working keymaps for the ANSI and ISO version of the Raise here: https://github.com/Dygmalab/chrysalis-api/tree/cafa064a7a680e3a41c7d4bf5f0508a3b22199c5/packages/chrysalis-hardware-dygma-raise/src/lib/components

mattvenn, 2019-05-30:
As I can't seem to link to discord messages, here's a discussion from the chrysalis group:
[image]

We already have the focus hook as described above. This should be called after the focus.help command to determine which keyboard is used.

https://github.com/Dygmalab/chrysalis-api/tree/master/packages/chrysalis-hardware-dygma-raise needs splitting into ANSI and ISO versions, and the correct one should be loaded depending on the answer from the hardware.

0957758592, 2019-06-26:
Look through reference [here](https://github.com/keyboardio/chrysalis-api/pull/22#issuecomment-505835316)

mattvenn, 2019-07-25:
this has now been fixed with 2 versions of our hardware and the new hardware.layout focus command

closed 2019-07-25 by mattvenn