keyboardio/Chrysalis #380: Add bossa firmware update

mattvenn, 2019-04-30:
Dygma Raise uses the SAMD21 chip from Atmel. The bootloader is based on the Arduino Zero, which uses bossa to upload new hex files.

mattvenn, 2019-06-04:
found a pure javascript implementation of sam-ba here: https://github.com/synthetos/node-sam-ba/blob/master/sam_ba.js
will test on the Raise

obra, 2019-06-04:
(license for this is GPL3. @algernon - how does that play with the license
regime for chrysalis?)
ᐧ

algernon, 2019-06-19:
chrysalis-api (and Chrysalis itself) is GPL-3-only, that plays nice with GPL3+, so as far as licensing is concerned, `node-sam-ba` is fine.

mattvenn, 2019-07-25:
we are now testing a fix for this - will submit a PR once we're happy

algernon, 2022-05-27:
We have recently removed Raise support from Chrysalis, so this issue is moot now. If/when we re-introduce support for the Raise (or any other keyboard that uses Bossa), we'll add the appropriate flasher too.

closed 2022-05-27 by algernon