keyboardio/Chrysalis #381: use a nice colour picker

mattvenn, 2019-04-30:
Use a react component colour picker and palette
[image]

TreTuna, 2019-05-02:
This is one that I definitely want to get in, but haven't found the best way to represent it on the UI yet. The current package we use does have the capability to switch over to one very similar to this though.

mattvenn, 2019-05-13:
@TreTuna , how about we make a mockup of where it would fit? If it looks good we can submit a pull request for the change.

algernon, 2019-05-13:
I'm more than happy to review both a mockup with a decent colorpicker, and a PR too. The current one was chosen because it was the easiest to fit into the space we had, and I lacked the time and skills to make use of a better picker provided by the react-color library (it provides plenty).

mattvenn, 2019-05-14:
Summary of main problems at the moment:

1/- confusing that palette buttons are split into 2 banks.
2/- the current picker is hex only
3/- there is limited space for a big colour picker

Our suggestion:

[image]

1/- colour picker is actually a button, that opens a popup with large fully featured colour picker

TreTuna, 2019-05-15:
Moving this to the side creates a bad UX. The user is already accustomed to the controls for key selection being on the bottom, it would feel weird to move that to the side when you switch to colors. This also creates a weird transition in the UI as you now have to squish and move over the keymap. The bottom works really well because it allows the keymap to get larger horizontally, which improves readability.

I do think that something with pop-ups will end up being the way to go for the color picker though. That's where my head was at the last time I really looked at this.

mattvenn, 2019-05-15:
So if we rotate this round and put it in the bottom we're good to go? Do you want to see another mockup or should we start making changes.

TreTuna, 2019-05-15:
The color groupings look decent enough to me. I don't like the "picker button" look personally as you can't tell what that is or what it does.

mattvenn, 2019-05-15:
I suggest we use something like this: 
[image]

So it's immediately obvious that it's the way to choose colours.

TreTuna, 2019-05-16:
These are the ones we have options with the current library we are using. The Chrome and Sketch versions are the closest to what you have pictured. I had also thought about using the Slider version as well. It could work well with the color groups on the left and the slider picker on the right. The picker could slide under the color groups if the window got smaller.

[image]

algernon, 2019-05-17:
How about having the palette laid out as @mattvenn proposed, but the color button being a real button, with an icon on it that shows the currently selected color? Clicking the button would pop up a dialog with a full-blown color picker (react-color's Chrome picker sounds like the best to me, that matches our design, and doesn't come with a built-in palette, which we don't need there).

algernon, 2019-05-17:
We'll have to roll our own palette implementation I think, like we do now, because the ones in `react-color` insist on using the color as the key, which breaks as soon as we have multiple instances of the same color (ie, the default blacks).

Perhaps we won't even need a color picker button, but could wire that up to right click? Though that's not very discoverable. So how about a palette layout very similar to react-color's Twitter picker, where we have spacing between the colors? The selected color would have a checkbox inside it, a'la the Swatches picker. And beside the palette, we'd have a palette picker icon-button.

mattvenn, 2019-05-17:
Hey, check this short video out to show our implementation. I like it a lot.
https://www.loom.com/share/ad6c45a910644dc8aa005c59bbdcf1fa

Roman is still waiting on the keyboard hardware to actually be able to implement this.

TreTuna, 2019-05-17:
That's not bad. I have one idea that I may be able to display this weekend as well. I'll be bringing my keyborad home tonight for the weekend so I'll have a little time.

algernon, 2019-05-17:
I like the video, except for the button. It looks like a color picker itself, rather than a button that pops up one. A floating action button with a color pick icon, perhaps with the icon having the same background as the current color - that'd be better in my opinion.

mattvenn, 2019-05-20:
How about this?
https://www.loom.com/share/9db6f8ae62ca40e48e44d285ee58be3b

algernon, 2019-05-20:
That looks better. Can you put a color picker icon in the middle of it, to make it super-clear that it's a button? (I'd just use a Floating Action Button with the selected color as background, to make it fit in well with the rest of the material UI)

mattvenn, 2019-05-20:
This one is cool
[image]

0957758592, 2019-05-20:
Here is the samples:

### VIDEO 1

https://www.loom.com/share/99acd0b11716447a917edf0507d3914a

Samples as in one line as double line

### VARIABLES to VIDEO 1

[image]
[image]
[image]

### VIDEO 2 (without separator)

https://www.loom.com/share/2b96c72e18304633b821681a2ab01118

algernon, 2019-05-20:
I like both, thanks for the demos! I think the second looks better, and I assume the picker comes up on right click, yes? That's great, but the separator with a button in the middle is more _discoverable_, probably. But I like both.

0957758592, 2019-05-20:
> I like both, thanks for the demos! I think the second looks better, and I assume the picker comes up on right click, yes? That's great, but the separator with a button in the middle is more _discoverable_, probably. But I like both.

picker could be implemented on any ways you would like with any variables e.g.:

1) 
- left click with ctrl/shift - will change the backgound color;
- single click will popup the colorPicker to change color button;

2) 
- left click - will cahnge the color;
- rigth click will open popup colorPicker;

... any variables

- also could add a button to reset to (any default color values);

mattvenn, 2019-05-20:
please don't go back to splitting the palette into 2. Have the 16 colours and then the picker. Otherwise it's really confusing. Users think the each block of 8 references each half of the split keyboard.

mattvenn, 2019-05-21:
[image]

* palette button is floating action button - plain, unchanging background
* clicking the palette button pops up react colour picker that edits the currently selected colour
* 16 colour buttons in one line, with highlight to show the one selected

0957758592, 2019-05-22:
### Here is a implementation:

https://www.loom.com/share/ce8152c0f4034f238e204dbfe0a36da7

But also have aquestion about min screens size buttons layout.
I didn't use flex yet here, but think it shoulbe implemented. **I case resizing screen.**

What the implementation of resizing there would be better? to divide it in two rows as here :

And the panelPicker button should be bigger

[image]

algernon, 2019-05-22:
Loving the new video, great work!

As for smaller window sizes, falling back to 2 rows would be the way to go in my opinion. Not sure about the bigger button though... perhaps vertically centered against the palette box instead?

0957758592, 2019-05-22:
### Here is a new responsive implementation :

https://www.loom.com/share/bbe36f8b713448c6a05523f54a43e6fa

### also could be looks like this (in the middle):

[image]

TreTuna, 2019-05-22:
I like the button to the right on smaller screen size much better. Having it split them feels... Gross.

I like @algernon's idea of not resizing, but just vertically centering the button on the right.

0957758592, 2019-05-23:
@TreTuna , @algernon 

### Here is a new implementation:

https://www.loom.com/share/58b00cc4cab749c6a5c2ee4fa8c4d353

TreTuna, 2019-05-23:
That looks great to me!

mattvenn, 2019-05-23:
OK, let's do it - can @0957758592 make the changes and make a pull request please.

0957758592, 2019-05-23:
> OK, let's do it - can @0957758592 make the changes and make a pull request please.

Yes sure!
I still waiting to the PCB to make a right functionality.

mattvenn, 2019-07-25:
can we close this now?

closed 2019-07-25 by obra