keyboardio/Chrysalis #382: multiple select LEDs

mattvenn, 2019-04-30:
allow control/shift select of keys to build selections and then assign them a colour.

TreTuna, 2019-05-04:
The method we had before that we've talked about bringing back was more of a painting feel. You select a color, then click each button you wanted to change. We lost this functionality when we combined the key and color editors together. I personally liked that version of the color editor a lot. I'd like to see what @algernon thinks on this one for direction.

algernon, 2019-05-13:
I agree with @TreTuna here, the paint approach was straightforward and easy to use. No need to hold shift or control either, or to pre-select keys: select the color, and paint like on a canvas. That's pretty much how paint programs work too (except our pixels are keys).

mattvenn, 2019-05-13:
sounds fine to me. So in this case do we revert something to a previous commit or is there work to be done? I'd like our new contributor Roman to handle this if possible - and he's new to the project. What's the best way of getting him up to speed on this?

algernon, 2019-05-13:
Can't really revert, because that'd revert the key editor-color editor merge too, so there's work to be done. I guess the easiest would be to check out an earlier version that had the two split, and the current one, so he can feel the difference. Looking at the code, or the merge (a031cb888fc26e5383aa05ee1f3b4318e8c7f6e7) in question might also be beneficial.

mattvenn, 2019-05-14:
OK, thanks. I think this is the way forward.

0957758592, 2019-05-31:
It will be better to realize it after merging #381 (newcolorPalette) to implement it.
Cause more part of functions will influence to realisation.

Now continue working of this relisation using new implementation in #392

0957758592, 2019-05-31:
The solution for this issue nearly finished. Couple changes left.
Waiting for merging PR #392 (to pull changes from master) and create new PR here to discuss implementation and resolve this issue.

0957758592, 2019-06-06:
Here is view of multiple select implementation issue #382:
how it works:
- select any key you want to change a color
- select a color button will be implemented with ctrl or shift
- to make sure you can make multiple selection the color-picker icon will appear inside the color-button
- just clicking any keys you want to change colors

https://www.loom.com/share/fc43078234d5420e93417ac0deaeaa0b

mattvenn, 2019-06-10:
I've just met with Roman, and we checked the pull request. There was some confusion about what we wanted because my original request was to use control/shift to multiple select.
We're now changing to the behaviour discussed above with @TreTuna and @algernon. Should get an update later today.

TreTuna, 2019-06-10:
Excellent! That was the one thing here I was a little wary on.

0957758592, 2019-06-11:
Here is a demo
https://www.loom.com/share/1806fd4eb42a4f0f931e0c95e2abdae7

But here is a confusing with clicking on the key:
1) to pick up (display) the bottom color-panel I click to the `Esc` key
2) when I click to the same key - the bottom color-panel should disappear

But, in case the `Esc` button is allocated - and I decide to change the color - bottom panel is disappear (and no any color-changes happens)

What the better way to go around?
Is would be better to change the function to disappear the bottom panel - just clicking outside the keys panel?
Or have to save current implementation? Trying something different?

0957758592, 2019-08-04:
It looks like this issue should be closed.
Solution in the project already.
[closed PR](https://github.com/keyboardio/Chrysalis/pull/398)

closed 2019-08-19 by obra