keyboardio/Chrysalis #384: reorganise colour picker and saved colours

mattvenn, 2019-05-03:
The saved colours should all be in one group. Having a separate group each side of the colour picker is confusing.

Recommend this layout as an example:
[image]

TreTuna, 2019-05-04:
This is the library we use for our color picker. http://casesandberg.github.io/react-color/

I had been looking at the _sketch_ version color picker on there for implementing with Chrysalis, though it takes up a lot of vertical space so the UI design would have to be changed for this.

TreTuna, 2019-05-04:
[image]

algernon, 2019-05-17:
Lets tackle this in #381, palette layout and the color picker are fairly tightly coupled, so I think one issue about them is enough, and #381 had more ideas laid out.

closed 2019-05-17 by algernon