keyboardio/Chrysalis #385: main window changes for usability

mattvenn, 2019-05-03:
We have done some user testing and suggest the following changes to improve usability.

[image]

TreTuna, 2019-05-04:
> 1. After changes made, the hamburger menu changes to a cross. Remove this functionality, and instead put a cross next to the save button. We already have a popup to discard changes.
I can get behind this, however, we also have to be ready for the case where the user has unsaved changes and clicks on the menu to navigate away from this screen. Currently, that's not possible because the clear functionality lives there. If we change that we have to edit the functionality of the menu when there are unsaved changes, either disabling it or setting up a warning modal if the user tries to navigate away.

> move the layer dropdown to the right, so it fits in the group of layer buttons
Agree completely here

> change the clear icon (a cross) to an erase icon
Material Design doesn't have an erase icon, however, I think the
[image] icon could work well here.

> remove keyboard name menu, put the menu items in the hamburger menu
I've thought about this as well, and I'll add to it where I'd like to put the keyboard name in the title of the window.

> never hide the bottom menu. Automatically select a key at launch and have this shown in the bottom menu.
I've gone back and forth with this. I personally like to hide the bottom menu a lot when I'm thinking about changes in my layout and looking at the color scheme.

mattvenn, 2019-05-14:
Hamburger menu changes to cross:
OK I understand your point. I think the problem here is how visually separated the save and cancel buttons are. So how about 
* keep the popup menu on the hamburger button
* don't change the hamburger menu to a cross
* add a cancel button next to save

Never hide the bottom menu:
In our testing we've seen users confused with not knowing what to do. Clicking a key to bring the menu up doesn't seem to be obvious. How about when the screen is first shown, a key should be autoselected so the menu or colour picking panel is present.

algernon, 2019-05-17:
> After changes made, the hamburger menu changes to a cross. Remove this functionality, and instead put a cross next to the save button. We already have a popup to discard changes.

Material UI discourages using two FABs, but it does say "Occasionally two FABs can be used, if they perform distinct, yet equally important, actions", which might apply in this case.

Mind you, the hamburger menu turns into a cross due to it being a [Contextual action bar](https://material.io/design/components/app-bars-top.html#contextual-action-bar), and I think this is a good thing. It makes it easier to stop the user from accidentally navigating away. We should, however, change the text of the appbar in this case, to make it clear that there are unsaved changes.

I hope that having the text change too on entering contextual mode would also make the purpose of the X clearer.

> move the layer dropdown to the right, so it fits in the group of layer buttons

This is a good idea, lets do that!

> change the clear icon (a cross) to an erase icon

Agree with @TreTuna, the `layers-erase` icon is a better fit.

> remove keyboard name menu, put the menu items in the hamburger menu

Material UI says that navigation items on the AppBar should be on the far left. I don't really see why moving them to the keyboard name menu would be any better. I kinda like that the keyboard name menu has links to various device-related sites, to be honest.

> never hide the bottom menu. Automatically select a key at launch and have this shown in the bottom menu.

I like to have the bottom menu hidden too, when thinking about the layout, to reduce any distraction. I do agree that it would be a better UX if we'd show it by default. For the short term, we can auto-select the first key on screen load. In the long term, I'd just show a generic bottom panel with controls disabled until a key's selected. Deselecting a key would then hide the panel.

...on second thought, perhaps that's not a particularly consistent behaviour, so auto-selecting on first load is probably a good longer-term solution too.

algernon, 2019-05-17:
Regarding auto-selecting a key: the problem there is that we need to look up the led index of the key we auto-select, because `key#0` might not be `led#0` (eg, `key#0` is `led#3` on the Model01). Shouldn't be too hard to do, but not trivial either.

For anyone embarking on this quest: you don't want to change the defaults in the constructor (because at first render, we don't have the keymap yet, so selecting the first element of an empty array will yield us a nice error), you want to set `currentKeyIndex` and `currentLedIndex` in `scanKeyboard()`.

Ptepp1c, 2019-05-18:
Regarding the menu at the bottom, I agree that I wasn't actually sure what to do at first since it wasn't there, perhaps have an option in the settings, default to always on, and a checkbox to make it behave how it does now for users who prefer to have it disappear.

0957758592, 2019-05-28:
Actually as I see - in component `Editor.js` there is the props `in` into  `<Slide in={this.getCurrentKey() != -1} direction="up" unmountOnExit>`  especially here: `in={this.getCurrentKey() != -1} ` that return `true` or `false` that activate or deactivate bottom panel. 

The better way to use already implemented functions - in that case as this props (without adding new features or functionalities).

It looks like: in case we will add props e.g. **inTrue**  here : 
`const { classes, in: inTrue } = this.props;`  after `render()`
and than will use it here: 
`<Slide in={inTrue || this.getCurrentKey() != -1} direction="up" unmountOnExit>`

it should display bottom panel, but will use the key as `-1`.

Its just a preorder to invokation, not as solution yet.
So in that case do we need to activate panel with the special key to display it as selected key:
e.g. in case app will run and in first present will appear the bottom panel with selected key as `ESC` (for example) or it should be appear without selected key but will return -1 ?

mattvenn, 2019-06-21:
closed because the issue tries to address too many things, some of which have already been fixed and merged. The bottom menu hiding part has been moved to https://github.com/keyboardio/Chrysalis/issues/401

closed 2019-06-21 by mattvenn