keyboardio/Chrysalis #386: reduce number of keytype groups

mattvenn, 2019-05-03:
At the moment there are many keytype groups. We suggest to reduce the number by grouping similar types of keys together. For example:

- Letters & Digits
- Punctuation & Spacing
- Navigation & misc
- shift to layer & lock to layer
- media & led
- numpad & fkeys
- mouse: move, button, wheel, warp
- one shot modify and layers

Multiple grouped items can have the same name. So we suggest grouping them with their names:

[image]

TreTuna, 2019-05-04:
I like the idea of combining some groups, however, there's another problem this brings up which also relates to current items on the docket for the 0.6.0 milestone. Currently, we already have a problem with there being too many things in that panel when the window is reduced in size. https://github.com/keyboardio/Chrysalis/issues/315 and https://github.com/keyboardio/Chrysalis/issues/210

Adding more stuff to this means we need to rethink the bottom panel a bit.

mattvenn, 2019-05-14:
Increase the window size? Decrease the keymap image?

algernon, 2019-05-17:
Decreasing the keymap image size quickly results in unreadable labels, I found. We can increase the minimum window size though, but not by much. I don't want to force it to `>= 1400px` or something like that.

There are certainly groups we could - and should - merge. Like all layer change buttons (`Lock`, `Shift`, `OneShot`) should be one group: `Layer change`. Mouse keys could fit in one group too. Basically, merging related smaller groups would make sense.

However, that will need a *lot* of code changes, due to the way the keyboard editor is currently built. Regrouping would be an incredibly beneficial task, for someone who's well versed in JS and isn't afraid of a big refactor. The current key grouping mechanism is *awful*, so if throwing it out the window entirely is the way to go, I won't be opposed to that either.

TreTuna, 2019-05-17:
Well, you know I'm not afraid of a big refactor. 😉 However, this section of code isn't even something I've looked at yet and I have no idea when I could get to it right now and think we need to work on simplifying a bit of the other code first. Now that I'm back at work I don't have my keyboard at home every night to dev on this.

algernon, 2021-01-25:
I think this has been addressed with the editor rework in 0.8.0.

closed 2021-01-25 by algernon