keyboardio/Chrysalis #388: All black caps and all black colour palette

nickhandel, 2019-05-14:
I couldn't see the colormap editor, so decided to update my firmware.  I found I couldn't update to 0.5.0 basic firmware (repeatedly got "error flashing firmware) so I chose the experimental firmware, which flashed OK.

I could now see the colormap editor, but the default shows all black caps, and all the colour palette blobs are black as well.

I have tried:

Removing all trace of the Chrysalis app (with an uninstaller tool) and re-installing
Resetting the EEPROM on the keyboard

Does this make any sense?  Is it somehow to do with a previous version of the separate colormap editor?

I had this issue previously with the original colormap editor.  I'm on latest Mojave.
[image]

Ptepp1c, 2019-05-15:
I have the same issue, however, as I am new to the Keyboardio I am not sure what the colour picker screen is meant to look like. So whether this is a bug or intentional let me know.

However, I found that you can still change the keyboard colours its just more finicky. 

I have seen this colour picker mentioned by tretuna http://casesandberg.github.io/react-color/ 

So thats what I used for hexcodes

Pick a colour you want from the site and it will tell you the hex code then you can assign them to the swatches at the bottom. The right most circle seems to be the base colour.

It is quite annoying to change all the key colours at the moment but I think thats being looked into.

One thing to note is that the default colour changes seems to remain (such as when you press numlock. Also, it seems to add an a new led colour option (the one you set up) rather than add to others, so it works well with solid colours, but cool patterns etc you will have to go with the defaults that come built in with the led key unless you program them yourself.

algernon, 2019-05-17:
This is both a bug, and not a bug. The default palette is all-black (which translates to LEDs being off), so an all-black palette and all-black keys on initial load is by design.

The color picker, however, is strange and unintuitive, we're tracking that in #381. The bug here is that the color picker is hard to use, and it's not exactly obvious how the colormap plugin works either.

Editing the colors won't immediately affect the keyboard, it will only do anything if you have the Colormap LED effect active, and as @Ptepp1c said, some plugins like the NumPad plugin will still override the colors. That's something we're working on on the firmware side, to make it less surprising.

algernon, 2022-05-17:
Because the palette being all-black on initial load is by design, and we have redesigned the palette and colormap editor since the issue was originally reported, I'm closing this.

closed 2022-05-17 by algernon