keyboardio/Chrysalis #391: Opening and closing Layer input/export window, then downloading messes up the keymaps.

mattvenn, 2019-05-20:
**Describe the bug**
Opening and closing Layer input/export window, then downloading messes up the keymaps.

Here is result of focus api calls to keymap.custom before and after bug

matt-pc:2038 [Kaleidoscope-v2-port]: cat before 
41 30 31 32 33 34 35 0 0 36 37 38 39 45 46 42 43 20 26 8 21 23 0 0 28 24 12 18 19 47 48 40 57 4 22 7 9 10 0 0 0 11 13 14 15 51 52 100 225 225 29 27 6 25 5 0 0 0 17 16 54 55 56 229 224 227 226 224 44 0 17409 17451 42 76 44 228 230 17152 17451 228 58 59 60 61 62 63 64 0 36 65 66 67 68 69 42 76 20 26 8 21 23 0 0 100 74 78 75 77 70 47 48 57 4 22 2838 9 10 0 0 0 11 80 81 82 79 52 40 225 29 27 6 25 5 6 0 0 0 17 16 54 55 56 229 224 17152 226 40 44 0 17451 65535 76 49 49 44 230 17152 118 228 65535 0 0 0 0 0 0 0 0 0 0 0 0 0 0 30135 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 41 30 31 32 33 34 35 0 0 36 37 38 39 45 46 42 43 20 26 8 21

matt-pc:2039 [Kaleidoscope-v2-port]: cat after 
58 58 59 60 61 62 63 64 0 36 65 66 67 68 69 42 76 20 26 8 21 23 0 0 100 74 78 75 77 70 47 48 57 4 22 2838 9 10 0 0 0 11 80 81 82 79 52 40 225 29 27 6 25 5 6 0 0 0 17 16 54 55 56 229 224 17152 226 40 44 0 17451 65535 76 49 49 44 230 17152 118 228 65535 60 61 62 63 64 0 36 65 66 67 68 69 42 76 20 26 8 21 23 0 0 100 74 78 75 77 70 47 48 57 4 22 2838 9 10 0 0 0 11 80 81 82 79 52 40 225 29 27 6 25 5 6 0 0 0 17 16 54 55 56 229 224 17152 226 40 44 0 17451 65535 76 49 49 44 230 17152 118 228 65535 65535 0 0 0 0 0 0 0 0 0 0 0 0 0 0 30135 1 1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 41 30 31 32 33 34 35 0 0 36 37 38 39 45 46 42 43 20 26 8 21

**To Reproduce**
Steps to reproduce the behavior:
1. Change to layer 1
2. Click the import/export layer button
3. Click 'ok' on the inport/export layer window popup
4. See that the top of the configurator is pink indicating there is a change (there is no change)
5. Click download button

**Expected behavior**
Nothing should change

**Screenshots**
Before clicking download button
[image]

After clicking download button
[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Chrysalis Version: master @6599799ec1d4cb374a88b13464adb0a0db54b810

mattvenn, 2019-05-20:
Related to https://github.com/keyboardio/Chrysalis/issues/348 ?

obra, 2019-07-17:
This should not have been resolved if the patch was applied to Dygma's fork, but not the core.

0957758592, 2019-08-04:
Is it should be closed?
The solution were been merged.

obra, 2019-08-04:
@danilov - Was the solution merged to keyboardio/Chrysalis? I see it as
merged to Dygma's branch.
ᐧ

0957758592, 2019-08-04:
> @danilov - Was the solution merged to keyboardio/Chrysalis? I see it as merged to Dygma's branch. ᐧ

@obra 
it seemd yes, 
looks like merged over there: https://github.com/keyboardio/Chrysalis/commit/0cc7412cb3c83a5de7f7cef3b789d3d72a23381d

**ScreenShot:**

[image]

obra, 2019-08-04:
> looks like merged over there: 0cc7412

Right. Things not merged to master should not be closed

closed 2019-08-19 by obra