keyboardio/Chrysalis #394: avast sees Chrysalis as a virus - hard to install on windows

mattvenn, 2019-05-30:
[image]

What do we need to do to make this easily install on Windows?

0957758592, 2019-05-30:
Im calling in AWAST directly now (+18443409251) to solve it to add our product into their whitelist.. Will provide a response from soon.

0957758592, 2019-05-30:
> Im calling in AVAST directly now (+18443409251) to solve it to add our product into their whitelist.. Will provide a response from soon.

Waiting for reply from avast withing 24 hours.

obra, 2019-05-31:
-Why- does it see it as a virus? (What does 'more detail' tell you?)
ᐧ

tsh-xx, 2019-05-31:
Symantec warned me that the executable had only been seen 4 times when I installed a few weeks ago. So it isn't (for me) detecting a problem, just being overly cautious - I don't remember the exact dialogue, but it was moderately scary. It was trivial to allow installation though.

mattvenn, 2019-07-01:
This is result of clicking on more details

[image]

[image]

It's easy to turn off the antivirus then install.
Does anyone know how this stuff works? Will each new version of the executable need to be added to the antivirus databases for this issue to be resolved?

algernon, 2021-01-25:
Perhaps signing the executable would get rid of the warning? Both Windows and macOS seem to be pushing hard towards signed installers. I suppose antivirus software would do likewise.

algernon, 2022-05-27:
I'm not sure we can do anything about it on our end, and we didn't have other reports of anti virus software causing issues, so I'm going to close this issue.

closed 2022-05-27 by algernon