keyboardio/Chrysalis #412: Can't flash firmware and manual flashing still prompts a firmware update

iamthebot, 2019-07-22:
**Describe the bug**
A clear and concise description of what the bug is.
Firmware flashing using release 0.6.0 on OSX 10.14.5 fails for both default and experimental firmware. Manual flashing of the chrysalis firmware bundle still causes Chrysalis to prompt for a firmware update.

**To Reproduce**
1. Launch Chrysalis on OSX
2. Click "connect" when the keyboard shows up on the list
3. Hold the PROG button and click to flash the firmware. Release PROG once it is glowing red
4. Firmware upload failed is displayed.

Verify permissions:
```
ls -l /dev/tty.usbmodemCkbio01E1
crw-rw-rw-  1 root  wheel   18,  52 Jul 22 13:31 /dev/tty.usbmodemCkbio01E1
```

My user is in the wheel group:
```
groups
staff wheel everyone localaccounts admin _appstore _lpadmin _lpoperator _developer _analyticsusers com.apple.access_ftp com.apple.access_screensharing com.apple.access_ssh
```

Next, I tried downloading the Chrysalis Firmware Bundle 0.6.0 release:

```
cd /Users/alfredo_luque/Documents/Arduino/Chrysalis-Firmware-Bundle-Chrysalis-Firmware-Bundle-0.6.0/Keyboardio/Model01/experimental
make flash
```

This time, the flashing process appears to complete successfully (for both default and experimental firmware):

```
make flash
BOARD_HARDWARE_PATH="/Users/alfredo_luque/Documents/Arduino/hardware" /Users/alfredo_luque/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash
Building ./Model01 0.0.0 into /var/folders/5w/2xytv5mj573clqspyyt6jqb40000gn/T//kaleidoscope-alfredo_luque/sketch/8637367249-Model01.ino/output...
- Size: firmware/Model01/Model01-0.0.0.elf
  - Program:   28042 bytes (97.8% Full)
  - Data:       1457 bytes (56.9% Full)

To update your keyboard's firmware, hold down the 'Prog' key on your keyboard,
and then press 'Enter'.

When the 'Prog' key glows red, you can release it.

2019-07-22 13:39:16.077 system_profiler[16056:111747] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-07-22 13:39:19.473 system_profiler[16065:111811] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44
```

Returning to the Chrysalis UI, it still prompts for a firmware update.

**Expected behavior**
Either firmware flashing should work on OSX when the correct permissions are set (oddly enough manual flashing through the terminal works) or Chrysalis should detect when the 0.6.0 firmware is already installed.

**Desktop (please complete the following information):**
 - OS: OSX 10.14.5
 - Chrysalis Version: 0.6.0

obra, 2019-07-22:
Is there more to the log of the manual firmware flashing? That log doesn't
actually show any flashing happen.

obra, 2019-07-22:
Somewhat reproduced here. The issue appears to be -detection- of current firmware on OSX.

https://www.dropbox.com/s/f5lufl25ppjjt51/Screenshot%202019-07-22%2015.09.53.png?dl=0 is a log of a couple detection attempts after verified-good firmware flashing on the linux side.

(I should note that we also had no problems doing the flashing from 0.6 on OSX. It just didn't get picked up by Chrysalis after we did.)

iamthebot, 2019-07-22:
@obra how can I generate more verbose logs of the flashing process both using chrysalis and via the Makefile?

obra, 2019-07-23:
We've reproduced it here, so hopefully you just need to wait for 0.61

algernon, 2019-07-24:
Reproduced, but I'm not quite sure yet when and how we broke things.

rkstar, 2019-07-27:
i just got my new m01 and i was hoping to just make the smallest change, swapping the cmd/shift keys on both thumb pads and move some media buttons around, so it can match my first m01. i'm also experiencing this problem with 0.5 and 0.6. hopefully 0.61 will be out soon!

algernon, 2019-07-27:
If you're seeing this with 0.5, then that's a different issue. What OS are you on? Does the user you're running Chrysalis with have write access to the device file (assuming OSX or Linux)?

rkstar, 2019-07-27:
i’m on mac os. i haven’t changed any permissions or done anything crazy. the device path in chrysalis is owned by root:wheel and my user is in the wheel group like the troubleshooting guide says.

closed 2019-07-28 by algernon

asaf-erlich, 2019-08-19:
@obra Even after testing on osx with 0.6.1 I'm unable to flash my firmware (or at least the GUI can't proceed). Should I open up a new github issue? There also aren't enough instructions to know how to take troubleshooting steps. I tried starting up chrysalis from command line. Has anyone else in this thread confirmed whether the fix worked for them?

rkstar, 2019-08-19:
@asaf-erlich -> i found that flashing did not work properly if my keyboard was plugged in to a usb hub. as soon as i plugged it directly into the computer it worked perfectly!

obra, 2019-08-19:
David: That's fascinating useful detail.  @asaf-erlich,  @algernon is
working on a new build of Chrysalis with significantly improved debugging
support. I'm hoping that will be ready in the next few days and will help
us get to the bottom of this.

Best,
jesse
ᐧ

asaf-erlich, 2019-08-21:
@rkstar I do have my keyboard plugged into a usb hub! I will try plugging it directly into my computer and see what happens.

@obra thank you I will definitely take advantage of those debugging options if they're available (and the suggested solution above does not fix my issue).

asaf-erlich, 2019-08-21:
@rkstar it worked! Thank you so much for the suggestion. Plugging directly into the computer successfully says it flashed my firmware and then I was able to connect and continue to try remapping of my keys.

obra, 2019-08-21:
@asaf-erlich - Once you have the 'right' firmware, can you then remap keys
using the keyboard plugged into the hub?

ᐧ

asaf-erlich, 2019-08-22:
Yes, @obra. I just tried it and once it's flashed it looks like I can remap keys even when it's plugged into the hub.

obra, 2019-08-22:
Thanks so much for the confirmation @asaf-erlich
ᐧ

obra, 2019-08-28:
> @obra <https://github.com/obra> does it make sense to reopen this and
> leave it open? Flashing still works correctly on linux even with a USB hub
> (and using the makefile on OSX).

It's actually worth testing 0.62, which may fix things.
ᐧ

algernon, 2019-09-09:
Out of curiosity, those of you who had problems flashing via a hub, when trying to flash, was the hub an unpowered or a powered one?

sbine, 2019-09-10:
@algernon I had problems flashing over a powered USB hub on 0.6.1, but just flashed successfully on 0.6.2.

rkstar, 2019-09-10:
@algernon i am using a powered usb hub

Mayeu, 2019-10-07:
I had this problem using an Apple USB-A -> USB-C adapter on my macbook. Using a USB-C <-> USB-C cable allowed me to flash the new firmware without issue.

commit 68998a57 (2019-07-28): Update the @chrysalis-api libraries to their latest

This should fix #412.