keyboardio/Chrysalis #415: Firmware update broken on OSX (0.6.1)

iamthebot, 2019-08-01:
**Describe the bug**
Firmware can be upgraded directly from the chrysalis-firmware-bundle using the provided makefile and `make flash` but firmware upgrade fails via Chrysalis. Tested with the experimental and normal firmware for 0.6.1.

**To Reproduce**
1. Open Chrysalis
2. Navigate to the firmware upgrade page
3. Select either firmware
4. Hold down the prog key and click the button to start
5. Release the prog key when it is glowing red
6. Observe that the firmware upgrade failed

**Expected behavior**
Firmware should successfully upgrade. Indeed, using the makefile works. I have also verified that the user is in the `wheel` group since tty.usbmodemCkbio01E1 is accessible by wheel.

**Desktop (please complete the following information):**
 - OS: OSX 10.14.5 beta
 - Chrysalis Version: 0.6.1

evanfields, 2019-08-01:
I am (unfortunately) reproducing this Chrysalis behavior, also 10.14.5/0.6.1.

obra, 2019-08-02:
@Alfredo - Once you do the manual update, does Chrysalis connect ok to the
keyboard?

iamthebot, 2019-08-02:
@obra Chrysalis as of 0.6.1 now recognizes the firmware flashed to the keyboard manually. This was not the case in 0.6.0 due to #412. The firmware not being recognized after a manual flash on OSX was fixed as part of #412 but firmware flashing not working via the Chrysalis UI has been the case since before 0.6.0.

iamthebot, 2019-08-02:
Also, I'm happy to provide diagnostics if you aren't able to reproduce this. Let me know how to dump additional logs if needed.

algernon, 2019-08-02:
If you can open the developer console (Menu -> Preferences -> Advanced -> Developer tools), try flashing then, and paste (or screenshot) whatever is on the JS console would be very useful.

plessbd, 2019-08-02:
[image]

iamthebot, 2019-08-06:
[image]

seeday, 2019-08-09:
I had this same issue and realized it was because I had another program (Cura) open that likes to steal serial ports when they're reset. Perhaps this is the issue?

iamthebot, 2019-08-09:
@seeday I don't think so because flashing via the CLI works (which I would expect wouldn't work if the ports weren't available otherwise).

mariusmorabosch, 2019-08-14:
I have the same problem here, cannot upgrade firmware via chrysalis

algernon, 2019-09-09:
When trying to flash, is the keyboard connected directly to the computer, or via an USB hub? If via a hub, is it a powered or an unpowered hub?

plessbd, 2019-09-09:
This is resolved for me with the 0.6.2 release

algernon, 2019-09-09:
Sweet, then this was a case of too short timers, and is different than #412 was. Can anyone else confirm this issue is gone with 0.6.2, just to be sure?

iamthebot, 2019-09-27:
Unfortunately this did not solve it for me (I'm on OSX Catalina). Now, It doesn't immediately crash but holding down the prog key never causes the keyboard to enter the flashing state (where red keys start appearing). CLI flashing via makefile still works so it's not a permissions issue.

My keyboard is connected to a 2018 15" MBP via a Satechi USB-C Pro Hub

obra, 2019-09-27:
can you test to see if it works without the hub involved?

dhosterman, 2019-11-01:
I can replicate @iamthebot's issue with or without a hub.

obra, 2019-11-01:
We've got confirmation that Apple broke flashing on Catalina. We're
expecting them to push a fix.
ᐧ

obra, 2020-03-06:
I believe this has been resolved. I am closing this ticket. If it’s still an issue please feel free to be open

closed 2020-03-06 by obra