keyboardio/Chrysalis #418: Arch Linux AUR Repo

iamthebot, 2019-08-07:
**Describe the solution you'd like**
Being able to install the latest Chrysalis via Arch Linux's User Repositories would be super nice. The AUR is well suited to alpha software since we can track git. Probably more so than applying to be in the official repos of any given Linux distro.

**Additional context**
I'm happy to start working on this. Hopefully others find it useful. Will update this issue when it's ready to go.

obra, 2019-08-07:
Awesome! Thanks!
ᐧ

iamthebot, 2019-08-07:
https://aur.archlinux.org/packages/chrysalis-git/ is ready and tracks master. Verified it installs correctly. Working on the non-git AUR package that pulls the latest AppImage now

iamthebot, 2019-08-07:
And the main binary release repo is ready too: https://aur.archlinux.org/packages/chrysalis/

closed 2019-08-07 by iamthebot

obra, 2019-08-07:
If you're up for submitting a README PR, I'd love a pointer to the package
for Arch users there.
ᐧ