keyboardio/Chrysalis #419: Optionally unlock previous layer when locking layer

Temmon, 2019-08-08:
**Is your feature request related to a problem? Please describe.**
When experimenting with alternate layouts, I seem to be unable to return to layer 0 if I lock to a higher layer without a firmware reset. This can cascade so that num lock and fn keys appear nonfunctional because may be shadowed by the higher layer. I can't tell if I'm having problems with that too. 

**Describe the solution you'd like**
On the lock layer to selection, include a toggle button that will unlock the current layer if selected so that one can revert to a lower layer. 

**Describe alternatives you've considered**
Setting the higher layer as a shift layer doesn't work because I want to keep holding a modifier key. I'm sure I can modify something in Kaleidoscope, but I would rather use the more user friendly Chrysalis. I can adjust the topmost layer through Chrysalis so it is transparent or reset the firmware to allow me to go back to layer 0, but those are clunky and the former requires resetting frequent imports and exports. 

**Additional context**
I'm new with Chrysalis and the Model 01, so I would welcome knowing how to lock and go back to full layers if I'm just missing something. As it is, if I map 1 to the physical A key on layer 4 and a button to change to lock to layer 0 somewhere else, lock to layer 4, then try to return to layer 0, I'll still be typing 1 instead of a. Not only that, I just tried doing that and I couldn't press A again until I shifted to layer 2 using the fn button and went back, which is just weird and probably its own bug.

gijja, 2019-08-13:
Setting a key to lock to layer is a toggle. That is, if you set a key to lock to layer n, when pressing the key a second time it will return to layer 0, provided that the toggle key in layer n is set to blank and transparent, or repeating the same lock to n. This works even if you are holding down a modifier key (or, at least, it worked for me).

Temmon, 2019-08-13:
Thanks for that. That works great. That behavior also probably helps explain the confusing things I saw when I was experimenting. Closing this.

closed 2019-08-13 by Temmon