keyboardio/Chrysalis #420: Per Key Selection of LED Effect

iamthebot, 2019-08-09:
**Describe the solution you'd like**
It would be nice to apply an LED effect on a per-key basis. For example, assigning "a" a green color and then selecting the "pulse" effect would cause the "a" key to slowly pulse green while on Layer 0.

Having a suite of LED effects packaged with Chrysalis would be super nice.

**Describe alternatives you've considered**
I could not use Chrysalis and just do this with the firmware, but it would be much nicer to do this via the GUI.

gijja, 2019-11-13:
I agree with this suggestion. Oddly enough, I made a "lock to" layer to create a couple of application specific key definitions. The lock to layer key that I defined made an LED pulse effect in blue while in the locked layer, nothing that I defined, it just did this, although I don't mind the random effect, it would be nice to be able to control it. Apparently this is possible.

algernon, 2022-05-27:
LED modes are by design full-keyboard only at the moment. Changing that would be a huge undertaking, and would require quite a bit of resources on the firmware side, and I don't see that happening anytime soon.

Do feel free to open a feature request against [Kaleidoscope](https://github.com/keyboardio/Kaleidoscope). If Kaleidoscope ends up with this feature, we can then improve Chrysalis too, but it needs to be in Kaleidoscope first.

closed 2022-05-27 by algernon