keyboardio/Chrysalis #423: Randomly repeating letters

framke, 2019-08-24:
I have a Keyboardio Model01, which I have been using with no problems for the past 6 months.

In the past week I have flashed the firmware from Chrysalis 0.6.0 and 0.6.2. Since updating the firmware, random different keys will repeat for 4 full lines or so. This happens every 5 or 10 minutessssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssss

I did not input the above. The problem just occurred. Using Windows 10.

obra, 2019-08-26:
Hi Framke,  Does this happen with the keyboard connected to another
computer as well? If you back down to 0.6.0, do you still see it?
ᐧ

antevens, 2019-12-30:
I'm seeing the same issue on a Model01 using 0.6.2, I also have an Ergodox (classic) and this does not seem to affect it with same firmware.

Seems to affect both Linux and Windows so it all points to the firmware, I believe it's somehow related to pressing multiple keys at the same time and releasing in a particular sequence, perhaps somehow related to modifiers.

I'll add more details if/when I can.

obra, 2020-01-02:
I'd love to know if backing down to an older firmware makes this go away.
ᐧ

framke, 2020-01-04:
I backed down to 0.5.0 yesterday and my repeated letters problem has disappeared.  

I would have done so much earlier but I was not able to flash the firmware previously. I suspect other Windows problems I was having (which have been resolved) were stopping my ability to flash the firmware.

Also, I tried several different USB ports as suggested in the threads on the Keyboard.io site, but that did not help.

algernon, 2021-01-25:
Does this still happen? If it does, are any of the keys that repeat dualuse keys? (as in, keys that do one thing when tapped, and act as modifier / layer switch when held)

antevens, 2021-01-25:
Wish I could help but my Model01 is on another continent.
My Ergodox remains unaffected.

framke, 2021-01-25:
Based on another thread I read, I tried replacing the ethernet cable between the two halves of the Model 01. This pretty much fixed the problem. 

I do have an issue (once every day or so) where the input I have typed does not show up on my display for 5 or 10 seconds... no keys will produce any effect during these frozen moments. I suspect that is different from the issue I originally posted about.

algernon, 2021-01-25:
Mhm. I'll close the issue for now. If it resurfaces (either for @antevens, @framke, or anyone else), please reopen it.

Thanks for the update!

closed 2021-01-25 by algernon

obra, 2021-01-25:
Can you keep track of which keys it is? Is it the same few keys? (This may be a hardware failure)