keyboardio/Chrysalis #426: Enable mapping Umlauts

MTrost, 2019-09-01:
In the letter key type group I can only find normal a-z letters. 

It would be a great help to be able to select special characters like umlauts.

gijja, 2019-09-04:
In MacOS using an American English keyboard layout, the umlaut is option-u then tap u and it is an umlaut. I assume the there are similar Windows and Linux key combinations. MacOS has all of the language specific characters available in similar combinations.

sonicbhoc, 2019-10-07:
Couldn't you also use the unknown keycodes option for that?

MTrost, 2019-10-10:
I tried using the unknown keycodes but wasn't able to make it work. I will have another look at it.

When I tried it last time I was able to get normal letters working with decimal codes in unknown keycodes, but not umlauts. Perhaps I used the wrong codes.

almereyda, 2019-11-29:
This is also important for me, to at least have a guide that explains how to setup keycodes used in the ´de_DE´ locale.

An idea would be to start with creating layer templates that we can import, plus providing PDF sheets to print where the keys actually lie.

This would mean Chrysalis is able to read and display the character map according to the operating systems locale :new:, and not only the predefined US layout, as keycode and character maps might be different, and to allow for exporting a vectorised version of a layer :new:. The latter appears more easy, as the screen version is already rendered as SVG.

Do these qualify as legitimate follow-up issues?

I would then to suggest renaming this issue to Internationalisation, and break it down into concise user stories for all anticipateable edge cases.

faxm0dem, 2020-09-18:
I also have this use-case for french glyphs

gulgi, 2020-11-15:
And for Swedish.
I just want to be able to write åäöÅÄÖ. I couldn't get the unknown keycode to work, sadly.

almereyda, 2020-11-16:
Maybe the @keyboardio developers could give us a pointer where in the code which kind of change need to be made to implement this. Then we could try ourselves in a community fork and try to forge a PR together?

faxm0dem, 2020-11-16:
I'm using the altgr-weur keyboard layout on Linux, and it works great without any changes to the firmware: just setup the compose key in the OS and then you can type all those umlauts, accents and whatnot

See https://choam.eu/altgr-intl/

gulgi, 2020-11-17:
> I'm using the altgr-weur keyboard layout on Linux, and it works great without any changes to the firmware: just setup the compose key in the OS and then you can type all those umlauts, accents and whatnot

Windows sadly doesn't have a compose key.

One option is to have Windows use a Swedish layout. But because Crysalis etc doesn't know about international keys .. I will have to create the layout "half blind". Would be very hard to do, and not portable at all. :/ 

Sending unicode characters (that apparently doesn't work 100%), or setting up some macros perhaps? Not sure what else I can do. :/

algernon, 2020-11-17:
Hi!

FYI, we're working on adding international key support to Chrysalis, so far FR-AZERTY is testable, and DE-QWERTZ is coming soon. You can find experimental builds in #591 - at the moment, it only supports normal keys only, layer keys and the rest are displayed, but can't be set yet. Those will come very soon aswell, though.

If you'd like to help out - and that'd be _much_ appreciated! -, [this](https://github.com/keyboardio/Chrysalis/blob/keyselector/104-redesign/src/api/keymap/ndb/fr/azerty.js) is how the FR-AZERTY codes look. It's basically a list of objects with a code and label overrides. The base table is [us-qwerty](https://github.com/keyboardio/Chrysalis/blob/keyselector/104-redesign/src/api/keymap/ndb/us/qwerty.js), the international mappings override those. Adding a new language is as "easy" as creating the override table, and modifying [ndb.js](https://github.com/keyboardio/Chrysalis/blob/keyselector/104-redesign/src/api/keymap/ndb.js#L31), so it imports the new table, and adds it to the known set of languages.

In the very near future, we'll have a small tool that can convert X11 keymaps to override files, which should make supporting almost any language trivial. Manual review, at least initially, will still be necessary, to make sure the tool works correctly. But even before the tool is ready, if anyone feels like having a go at it, #591 is where you should begin.

faxm0dem, 2020-11-18:
Awesome, thanks @algernon and the link to `azerty.js` is broken

algernon, 2020-11-18:
It is, because @obra found us a better way to do it last night: using the [Unicode CLDR](http://cldr.unicode.org/index/charts/keyboards) project. I have updated #591 to use the keyboard layouts from that project, and we have FR-AZERTY, DE-QWERTZ and HU-QWERTZ available in the latest build there.

In other words, we no longer need to manually write overrides, we'll just use the XMLs provided by the CLDR project. All we need to do is copy the appropriate file over, and tell Chrysalis to use them.

algernon, 2020-11-18:
Preview of the current state:

[image]

algernon, 2021-01-25:
I think this is now addressed with the 0.8 release. There's probably a few things to iron out, but the bulk of it should be available on master, and in the latest release.

closed 2021-01-25 by algernon

almereyda, 2021-01-26:
Very much appreciated. Will report back from testing some time.