keyboardio/Chrysalis #429: Cannot open Chrysalis on macOS Mojave with stricter-than-default security settings

imrekoszo, 2019-09-13:
**Describe the bug**
When macOS is set to only allow apps from the app store or identified developers, Chrysalis won't be allowed to start.

**To Reproduce**
Steps to reproduce the behavior:
1. Set macOS to only allow apps from the app store or identified developers
2. Install Chrysalis (0.6.2) either via dmg download or brew
3. Try to launch it
4. See error as in screenshot

**Expected behavior**
Chrysalis launches.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS:   System Version:	macOS 10.14.6 (18G95)  Kernel Version:	Darwin 18.7.0
 - Chrysalis Version: 0.6.2

obra, 2019-09-13:
Put another way, Chrysalis isn't a signed app on OS X.
ᐧ

imrekoszo, 2019-09-15:
Thanks for the clarification, I'm an OSX noob.

obra, 2019-09-15:
(Just as a note, this isn't likely something we'll fix in the near future
unless macOS forces us to by further locking down their defaults.)
ᐧ

imrekoszo, 2019-09-17:
That's fine, I'm just trying to use it on a work machine where I'm not the one defining security. Workaround: run from source :p

brow, 2019-10-10:
If you open the app by right clicking it and selecting Open, then the dialog you screenshotted will still appear but will contain an additional button that allows you bypass this restriction.

imrekoszo, 2019-10-11:
> open the app by right clicking it and selecting Open

Damn dude, thanks very much, it does work indeed! This method should be more widely know.

focusaurus, 2020-07-27:
@brow FYI on macos catalina 10.15.6, even opening from finder with the context menu does not seem to allow running.

[image]

obra, 2020-07-27:
We absolutely need to fix this, but command-clicking and clicking “open” does seem to work for me on current Catalina. 

-Jesse

focusaurus, 2020-07-28:
I'm on a work-IT-managed mac so maybe it's extra locked down, but command-left-click does not show any menu. I also tried command-left-click on the "Open" context menu item itself but still no dice. My workaround is I have to run Chrysalis on a nearby linux laptop and swap the USB cable to tweak my layout.

kevinr, 2020-07-28:
That is definitely something we locked down at previous jobs 😰

obra, 2020-07-29:
Can you folks try
https://kaleidoscope-builds.s3.us-west-2.amazonaws.com/Chrysalis/255/Chrysalis-0.7.9%2B255.dmg

It should now be signed and notarized. 0.8.0 will be the first "release"
that's signed.

focusaurus, 2020-07-29:
It works! I'm able to launch it and run OK. Thanks!

obra, 2020-07-29:
Awesome. I'm so happy to hear it. I'm going to close this ticket.

closed 2020-07-29 by obra