keyboardio/Chrysalis #430: Can't change colors on layer 1 right half

darinclarkdev, 2019-10-04:
Chrysalis will pretend it works but as you can see it doesn't. The left half will accept color changes and some blank keys on the right half. The left half also doesn't accept key mappings in layer 1.
[image]

obra, 2019-10-04:
I believe that what you're running into isn't "the right half" so much as
"the numpad layer's numpad keys". Is that right?
ᐧ

darinclarkdev, 2019-10-04:
It is any key on layer 1. If you happen to be locked on layer one when you save changes some of the colors will temporarily work. As soon as you switch to layer 0 and back to layer 1 they are all red again.

sonicbhoc, 2019-10-07:
I have the same problem. but I can't even get it to temporarily work on mine.

tmarzolf, 2021-01-19:
I ran into something similar on Layer 1 - red keys that I couldn't change the color. I ended up abandoning Layer 1...

algernon, 2021-01-25:
This is, indeed, the Numpad plugin overriding custom colors. We'll need to figure something out about that.

algernon, 2022-05-27:
In the not too distant future, we'll be updating the firmware so that it ships with a built-in colormap that renders the separate Numpad plugin obsolete. Once that happens, this issue will stop being a problem.