keyboardio/Chrysalis #434: One shot modifiers does not work out of the box

Nimamoh, 2019-10-17:
The keys defined as one shot modifiers are just disabled.

Steps to reproduce the behavior:
1. Download last release of chrysalis. (currently the 0.6.2)
2. Open chrysalis, say yes when it ask to flash the firmware.
3. Define shift as a one shot modifier (for example)
4. shift now doesn't do anything

**Expected behavior**
The key defined as a one shot modifier should behave as such and keep its normal function when using it in a classic chord combination.

**Desktop (please complete the following information):**
 - OS: Windows
 - Chrysalis Version: 0.6.2

algernon, 2019-10-17:
When asked to flash the firmware, did you flash the "default, minimal" one, or the "experimental" one? Only the latter contains support for OneShot modifiers.

(This should be better explained somewhere, mind you)

Nimamoh, 2019-10-18:
Indeed I flashed the minimal one. I should have read further before asking.

closed 2019-10-18 by Nimamoh