keyboardio/Chrysalis #435: One shot modifier on shift key produces a (

joelcahalan, 2019-10-17:
**Describe the bug**
OneShot not behaving as expected

**To Reproduce**
Steps to reproduce the behavior:
1. Install experimental firmware/switch to layer three being default.
2. modify left shift to use oneshot
3. fast tap left shift
4. See "(" instead of the expected behavior

**Expected behavior**
Expect OneShot behavior to work(ie: fast tap will capitalize the next key typed)

**Desktop (please complete the following information):**
 - OS: mac10.14.6
 - Chrysalis Version: 0.6.2

gijja, 2019-11-17:
You probably have Space Cadet enabled on the left shift. Try going to space cadet with the shift key selected, select  disable Space Cadet, save - the key will lose its definition - so go to modifiers, assign left shift to the key, go to One Shot Modifiers, enable on the key and it should work. I tried this several times and it is consistent. Space Cadet seems to override OSM, which makes sense to me, although without documentation it is confusing. Also, when you go to the Space Cadet submenu, it doesn't indicate whether the particular key is enabled or not. Thus, you can have Space Cadet enabled without realizing it.

joelcahalan, 2019-11-18:
@gijja I tried what you describe with no success. I still get the paren as long as the experimental firmware is installed...and oneshot does not work at all for me.

gijja, 2019-11-19:
It seems strange that we would get

gijja, 2019-11-19:
It seems strange that we would get different results that neither of us can replicate. One question, did you ever explicitly invoke Space Cadet for your left shift key? Thinking back, I had Space Cadet on my shift key involuntarily when I updated Chrysalis from .5.0 to .6.2, I wanted to eliminate it, and eventually I explicitly assigned Space Cadet to the key, saved, exit Chrysalis, opened Chrysalis and disabled Space Cadet (but eventually enabled it because I found it useful).

javierv, 2019-12-10:
Similar issue here, although I don't use OneShot but my OS's (Linux) "sticky keys" functionality.

It works fine with 0.6.2 but since with that version the "Modifier when held, normal key otherwise" functionality isn't working for me, I've upgraded to the experimental version.

By the way, in my case the left shift generates  ")", and the right shift generates "=".

javierv, 2019-12-10:
Oh, sorry, I've just realized this issue seems to be a duplicate of issue #427 :thinking:.

algernon, 2021-01-25:
Indeed, this is a duplicate of another issue.

closed 2021-01-25 by algernon