keyboardio/Chrysalis #436: macOS 10.15 crashes when waking from sleep

JonathanTroyer, 2019-10-22:
**Describe the bug**
When resuming from sleep, macOS 10.15 crashes and restarts if a Keyboardio Model 01 (firmware version 0.6.2 default) is plugged in

**To Reproduce**
Steps to reproduce the behavior:
1. Plug a Model 01 into a Macbook (mine is a 15" 2015 Pro)
2. Let the computer sleep
3. Attempt to wake the computer

**Expected behavior**
The mac should resume properly

**Screenshots**
```
Sleep transition timed out after 180 seconds while entering darkwake on way to sleep. Suspected bundle: com.apple.driver.usb.AppleUSBXHCI. Thread 0x113fc.
Failure code:: 0x43f6e23f 00000031

================================================================
Date/Time:        2019-10-22 15:27:06 -0500
OS Version:       ??? ??? (Build ???)
Architecture:     x86_64
Report Version:   29

Data Source:      Stackshots
Shared Cache:     0x3b08000 9E444A44-3839-31D4-B170-609508E90F08

Event:            Sleep Wake Failure
Duration:         0.00s
Steps:            1

Time Awake Since Boot: 12000s

Process:          kernel_task [0]
UUID:             70EDD61F-86EE-3E1B-873F-98D909B78160
Architecture:     x86_64
Version:          Darwin Kernel Version 19.0.0: Wed Sep 25 20:18:50 PDT 2019; root:xnu-6153.11.26~2/RELEASE_X86_64
Footprint:        359.18 MB
Start time:       2019-10-22 15:27:06 -0500
End time:         2019-10-22 15:27:06 -0500
Num samples:      1 (1)

  Thread 0x6a    Thread name "daemon.thread-exception"    1 sample (1)    priority 80 (base 80)
  <IO tier 0>
 *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 942048) [0xffffff80002e5fe0] 1

  Thread 0x6b    1 sample (1)    priority 95 (base 95)
  <IO tier 0>
 *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 942398) [0xffffff80002e613e] 1
   *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 1659231) [0xffffff800039515f] 1
     *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 1440543) [0xffffff800035fb1f] 1
       *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 1446708) [0xffffff8000361334] 1
         *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 2504776) [0xffffff8000463848] 1

  Thread 0x6c    1 sample (1)    priority 95 (base 95)
  <IO tier 0>
 *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 942398) [0xffffff80002e613e] 1
   *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 1659231) [0xffffff800039515f] 1
     *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 1440543) [0xffffff800035fb1f] 1
       *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 1446708) [0xffffff8000361334] 1
         *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 2504776) [0xffffff8000463848] 1

  Thread 0x6e    1 sample (1)    priority 95 (base 95)
  <IO tier 0>
 *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 942398) [0xffffff80002e613e] 1
   *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 2413012) [0xffffff800044d1d4] 1
     *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 1440543) [0xffffff800035fb1f] 1
       *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 1446708) [0xffffff8000361334] 1
         *1  ??? (<70EDD61F-86EE-3E1B-873F-98D909B78160> + 2504776) [0xffffff8000463848] 1

  Thread 0x6f    Thread name "IOServiceTerminateThread"    1 sample (1)    priority 81 (base 81)
[... 381 lines cut ...]
Model: MacBookPro11,4, BootROM 195.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 2.2 GHz, 16 GB, SMC 2.29f24
Graphics: kHW_IntelIrisProItem, Intel Iris Pro, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533642465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533642465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x152), Broadcom BCM43xx 1.0 (7.77.105.1 AirPortDriverBrcmNIC-1429)
Bluetooth: Version 7.0.0f8, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
PCI Card: pci1b21,1142, sppci_usbxhci, Thunderbolt@192,0,0
PCI Card: Apple 57762-A0, sppci_ethernet, Thunderbolt@196,0,0
Serial ATA Device: APPLE SSD SM0256G, 251 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Bluetooth USB Host Controller
USB Device: Model 01
USB Device: USB 3.0 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
Thunderbolt Device: 34UM88, LG Electronics, 1, 25.1
```

**Desktop (please complete the following information):**
 - OS: macOS 10.15
 - Chrysalis Version: 0.6.2

**Additional context**
Not that I'm aware of. Let me know if this should be filed in [Kaleidoscope](https://github.com/keyboardio/Kaleidoscope) instead.

obra, 2019-10-23:
Huh. We've poked some folks on this one..This...should not be possible to
do to a mac from an external keyboard. At the same time, we've got a bunch
of other folks running Model 01s with similar firmware versions against
Catalina and they're not seeing this issues.
ᐧ

JonathanTroyer, 2019-10-24:
I think I've actually pinned it to my LG 34UB88-P, which, incidentally, won't work with my Model 01.

closed 2019-10-24 by JonathanTroyer

obra, 2019-10-31:
@JonathanTroyer Can you email me at jesse@keyboard.io? I've got a line on getting the LG issue fixed and would like to introduce you in private mail.