keyboardio/Chrysalis #438: Flashing broken in macOS Catalina

gibfahn, 2019-11-19:
**Describe the bug**
I can't manage to update the firmware on macOS Catalina (10.15.1).

**To Reproduce**
Steps to reproduce the behavior:
1. Click `Update Firmware`
2. Hold down `Prog`
3. `Update` button spins endlessly

**Expected behavior**
Flashing to succeed.

**Screenshots**
If applicable, add screenshots to help explain your problem. If possible, please also include a log from the "developer console". You can access it by pressing `Control + Shift + I`. When it appears, right click inside it, and select "Save as", and attach the saved file to this report.

**Desktop (please complete the following information):**
 - OS: macOS
 - Chrysalis Version: 0.6.2 (`brew cask install chrysalis`)

[image]

**Additional context**

I note this comment from Jesse (https://github.com/keyboardio/Chrysalis/issues/415#issuecomment-548929273):

>We've got confirmation that Apple broke flashing on Catalina. We're
expecting them to push a fix.

I'm guessing this is my issue, just raising this so there's an obvious issue for other users to notice if they (like me) buy a Keyboardio and get confused that Chrysalis doesn't work out of the box.

obra, 2019-11-19:
> Maybe Chrysalis could detect that you're running Catalina and show a
> warning message to say that there's a known bug (and even link to this
> issue or some page from Apple covering this).

As of now, Apple hasn't said anything publicly, but we expect it to be
fixed in 10.15.2, which will likely ship before the next Chrysalis release.
ᐧ

gibfahn, 2019-11-19:
@obra also linked to the upstream Arduino issue, which has more details: https://github.com/arduino/Arduino/issues/9290#issuecomment-542728021

MarqueIV, 2019-11-20:
> As of now, Apple hasn't said anything publicly, but we expect it to be fixed in 10.15.2, which will likely ship before the next Chrysalis release. ᐧ
> […](#)

Can you comment on where you're hearing this will be addressed in 10.15.2?  AFAIHS, Apple has been pretty silent on this. Would be great if they acknowledged what's going on. (BTW, I'm the one that opened the other issue referenced here.)

obra, 2019-11-20:
Unfortunately, I can't, though I'm pretty confident. (And given that Apple
has been silent elsewhere, I'm going to edit the public version of these
comments to be a bit more circumspect.)
ᐧ

codymullins, 2019-12-03:
Hey so, I am running Catalina on a 16" and it works perfectly fine from what I can tell. Might be isolated to pre-16" models?

obra, 2019-12-03:
@Cody - What subrev of Catalina are you running? It may well be that the
16" is running a prerelease of .2.
ᐧ

thebitguru, 2019-12-11:
Yesterday, with 10.15.1, I was having the same issue.  10.15.2 was released today so I upgraded, and just successfully installed the Chrysalis firmware without any tinkering in macOS.  So, it seems like this problem is fixed in 10.15.2.

codymullins, 2019-12-11:
@obra - I'm sorry, I just realized that remapping the keyboard is not the same as flashing. I'm on 10.15.1, haven't tried actually flashing. I'll try today.

obra, 2019-12-11:
10.15.2 is out and is reported to fix it. 

Beat, Jesse

closed 2020-01-14 by algernon