keyboardio/Chrysalis #445: Atreus2: Can't flash

obra, 2019-11-21:
Connecting to /dev/ttyACM0
App.js:150 Probing for Focus support...
FirmwareUpdate.js:151 ReferenceError: port is not defined
    at Object.flash (/tmp/.mount_Chrysa2oa3xe/resources/app.asar/node_modules/@chrysalis-api/hardware-keyboardio-atreus2/lib/chrysalis-hardware-keyboardio-atreus2.js:58)
    at Object.flash [as _flash] (webpack:/src/renderer/screens/FirmwareUpdate.js:142)
    at Object._flash [as onClick] (webpack:/src/renderer/screens/FirmwareUpdate.js:149)
Qe @ FirmwareUpdate.js:151

is what I get in the UI.

algernon, 2019-11-22:
This one I already fixed in [chrysalis-api](https://github.com/keyboardio/chrysalis-api/commit/980433abf06093de0353d31ef3f41667521e6154), just need to update the dependencies.

algernon, 2019-11-22:
Packages are updated on master, closing this.

closed 2019-11-22 by algernon