keyboardio/Chrysalis #452: `yarn install` doesn't work on Arch Linux

skeet70, 2019-12-10:
**Describe the bug**
`yarn` can't install dependencies on Arch because the `serialport` dependency won't build.

**To Reproduce**
Steps to reproduce the behavior:
1. Be on Arch
2. Clone this repo
3. Run `yarn`
4. See error

**Expected behavior**
All dependencies are built and installed, leaving the project in a state that `yarn test` and `yarn start` will work.

**Screenshots**
https://gist.github.com/skeet70/9b211e44fbde5cf19bdf9a4ef2455230

**Desktop (please complete the following information):**
 - OS:
    ```console
    λ ~/code/oss/Chrysalis/ fix-default-imports hostnamectl
       <snip>
       Operating System: Arch Linux
            Kernel: Linux 5.3.7-arch1-1-ARCH
            Architecture: x86-64
    ```
 - Chrysalis Version: [git master](https://github.com/keyboardio/Chrysalis/commit/12e938ae3bff7d1a6cc3f29c7fadc8b7baa40b18)

**Additional context**
Running `yarn build:linux` even without `yarn` having completed succeeds (including a section about building native serialport), and then `yarn start` seems to use those files somehow and starts up a local version of the application. `yarn test` still doesn't work.

skeet70, 2019-12-10:
Renaming this issue, it's not Arch specific, it's Node 12.

```console
λ ~/code/oss/Chrysalis/ master* nvm use --lts
Now using node v10.16.0 (npm v6.10.3)
λ ~/code/oss/Chrysalis/ master* rm -rf node_modules 
λ ~/code/oss/Chrysalis/ master* yarn
yarn install v1.21.0
$ node tools/preinstall.js
[1/4] Resolving packages...
[2/4] Fetching packages...
info fsevents@1.2.7: The platform "linux" is incompatible with this module.
info "fsevents@1.2.7" is an optional dependency and failed compatibility check. Excluding it from installation.
info fsevents@1.2.9: The platform "linux" is incompatible with this module.
info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
[4/4] Building fresh packages...
$ electron-builder install-app-deps
Configuring yargs through package.json is deprecated and will be removed in the next major release, please use the JS API instead.
Configuring yargs through package.json is deprecated and will be removed in the next major release, please use the JS API instead.
  • electron-builder version=20.40.2
  • loaded configuration file=package.json ("build" field)
  • loaded parent configuration file=/home/mumu/code/oss/Chrysalis/node_modules/electron-webpack/out/electron-builder.js
  • rebuilding native production dependencies platform=linux arch=x64
  • rebuilding native dependency name=serialport
  • rebuilding native dependency name=usb
Done in 24.79s.
```

skeet70, 2019-12-10:
Still can't run tests though:

```console
λ ~/code/oss/Chrysalis/ master* yarn test
yarn run v1.21.0
$ ELECTRON_RUN_AS_NODE=true ./node_modules/.bin/electron ./node_modules/.bin/jest
 FAIL  src/renderer/components/__tests__/DeviceMenu.test.js
  ● Test suite failed to run

    Cannot find module '../DeviceMenu' from 'DeviceMenu.test.js'

      20 | import "jest-dom/extend-expect";
      21 | 
    > 22 | import DeviceMenu from "../DeviceMenu";
         | ^
      23 | 
      24 | afterEach(cleanup);
      25 | const mockOpenMenu = jest.fn();

      at Resolver.resolveModule (node_modules/jest-resolve/build/index.js:230:17)
      at Object.<anonymous> (src/renderer/components/__tests__/DeviceMenu.test.js:22:1)

Test Suites: 1 failed, 1 total
Tests:       0 total
Snapshots:   0 total
Time:        0.897s
Ran all test suites.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

skeet70, 2019-12-10:
[Related tracking issue](https://github.com/serialport/node-serialport/issues/1742)

Looks like a newer version of `serialport` would work with Node 12. 

```console
λ ~/code/oss/Chrysalis/ master* yarn why serialport
yarn why v1.21.0
[1/4] Why do we have the module "serialport"...?
[2/4] Initialising dependency graph...
[3/4] Finding dependency...
[4/4] Calculating file sizes...
=> Found "serialport@6.2.2"
info Reasons this module exists
   - "@chrysalis-api#focus" depends on it
   - Hoisted from "@chrysalis-api#focus#serialport"
   - Hoisted from "@chrysalis-api#hardware#@chrysalis-api#hardware-softhruf-splitography#@chrysalis-api#flash#avrgirl-arduino#serialport"
info Disk size without dependencies: "820KB"
info Disk size with unique dependencies: "1.75MB"
info Disk size with transitive dependencies: "6.03MB"
info Number of shared dependencies: 73
Done in 0.69s.
```

Also looks like some upstream deps both in and out of Chrysalis' control would need to be updated.

qayshp, 2019-12-17:
Have been having similar problems building on macOS for a few months, but didn't report it because I didn't know if it was just my env.
Since I use homebrew and autoupdate to latest, I'm on node v13.3.0.

TreTuna, 2019-12-19:
I'll do some testing on Node 12 issues this weekend and see what I can come up with.

algernon, 2020-01-01:
I've done a quick update to the latest serialport package, and while everything compiled without even a warning, Chrysalis was no longer able to find keyboards. I didn't dig deeper, but it looks like simply bumping the dependency will not be enough.

algernon, 2020-02-12:
This has been addressed on master a few days ago. \o/

closed 2020-02-12 by algernon