keyboardio/Chrysalis #455: add windows 8 usb support

yangju2011, 2019-12-31:
Hi I just got a new keyboard and then realize that it only supports Windows 10, but I have Windows 8. I wonder what is the alternative way to make it work without reinstalling a Windows 10 OS on my computer.

obra, 2020-01-02:
Hi Ju,

I'm so sorry.

Your keyboard itself should work just fine on Windows 8. I believe the
issue you're running into is just with the configuration tool.

We don't have a whole lot of customers running Windows 8. What's going on
is that in Windows 8, Microsoft added a signed driver requirement for
Serial Port drivers, which they then provided a better option for in
Windows 10. Because Microsoft  ended mainstream support for Windows 8.1 (
https://support.microsoft.com/en-us/help/13853/windows-lifecycle-fact-sheet)
before
we released Chrysalis, Windows 8 has...not been much of a target.

The manual driver installation instructions are available here:
https://github.com/keyboardio/Kaleidoscope/wiki/Installing-USB-Driver-on-Windows
While I can't 100% promise that they'll resolve things for you, I believe
that they will and they're definitely worth a shot.

Best,
jesse

ᐧ

yangju2011, 2020-01-03:
thanks for the support jesse. i tried the driver installation, but it seems the keyboard i have does not work with the driver. thanks all the same.

on the other hand, i realize that i can configure my keyboard on other people's computer that has a windows10 OS. all the configs are stored. so i guess it is not that big a problem for me :)

closed 2020-01-03 by yangju2011