keyboardio/Chrysalis #458: Merging Preferences and Keyboard Settings

algernon, 2020-01-14:
**Is your feature request related to a problem? Please describe.**

Right now, we have two kinds of preferences: the `Preferences` menu, and a `Keyboard Settings`. While it makes some sense they're separate, because one affects the preferences of the application itself, and the other is for settings stored on the keyboard (mostly...), it is still a bit unconventional to have two preferences menus.

**Describe the solution you'd like**

Ideally, we'd merge the two (screenshot from Bazecor):

[image]

The keyboard-related settings wouldn't show until connected to a keyboard. This way, we'd have all settings in one place.

**Describe alternatives you've considered**

The two alternatives I considered were leaving things as-is, but I do not think that's our best interest; and the other was moving the two screens closer in the hamburger menu, but that would make it look awkward, and wouldn't really solve the core issue of having two preferences pages.

**Additional context**

The merging of the two screens was already done for Bazecor, and most of the commit can be lifted as-is. But before doing that, I'd rather ask for opinions first.

obra, 2020-01-14:
I'm ok with it, but I feel like they should be broken out into separate
tabs or panes or sections or something.

-j

>

algernon, 2020-01-14:
They're in separate cards. You can't easily see that on the screenshot, because in Bazecor, the two settings we had in Preferences (language & dark mode) were removed temporarily, so the not connected state only shows the "Advanced" accordion. For Chrysalis, there'd be a "Preferences" card above "Keymap settings", in a similar setup: title outside of the card, then the card itself.

The same happens for the advanced settings, where devtools and verbose logging (on our current Preferences screen) are in a separate card than the eeprom reset (on our current keyboard settings screen). *This* can be seen on the screenshot. The non-advanced settings would be similarly separated.

obra, 2020-01-14:
I'm fine with trying that.

algernon, 2020-01-14:
Ok, I'll push a PR up later tonight or tomorrow then, with screenshots attached.

Thanks!

PR #461 (2020-02-12): Merge the KeyboardSettings menu into Preferences
[image]
[image]

Fixes #458.
closed 2020-02-12 by obra