keyboardio/Chrysalis #459: Keyboard detection not working on Gentoo Linux

edannenberg, 2020-01-26:
**Describe the bug**
Model 01 Keyboard not detected on Gentoo Linux

**To Reproduce**
Steps to reproduce the behavior:
1. Start Appimage on Gentoo Linux
2. Scan for keyboards shows `No keyboards found!`

**Expected behavior**
Model 01 Keyboard is detected

**Desktop (please complete the following information):**
 - OS: Gentoo Linux
 - Chrysalis Version: 0.6.2+132

**Additional context**
dmesg and the shell the appimage was started from show no errors at all. The keyboard itself is working and detection/mapping via Chrysalis works on Windows on the same box (dual boot). I run a fairly minimal kernel so it's likely I'm missing a required option, it's impossible for me to tell though.

algernon, 2020-01-26:
Does the user you run Chrysalis with have permission to access the serial port of the Model01? (`/dev/ttyACMx`)

Or, since you're running a minmal kernel: does the serial port even show up? If not, you may need USB serial support in the kernel.

edannenberg, 2020-01-26:
Thanks for the swift reply! Adding `CONFIG_USB_ACM=y` resolved the detection issue. When trying to connect the keyboard the app now states that a firmware update is needed. I already did the firmware update when trying this on windows. Is this normal when switching the OS or might there be another issue?

algernon, 2020-01-26:
Then it's either permissions, or something (like ModemManager) interfering.

edannenberg, 2020-01-26:
I added my account to the `uucp` group so it should have had permissions, maybe it wasn't active yet. After a quick `chown` on the device to my account everything seems to be in order now. Thanks!

Would be nice if the app would throw proper error message for these cases:

* `/dev/ttyACM` is missing => ensure `CONFIG_USB_ACM` is active
* the current user account doesn't have the required permissions to access the device

algernon, 2020-01-26:
Yeah, better error messages is something we're working on. (See #276, for example)

closed 2020-01-26 by edannenberg