keyboardio/Chrysalis #463: You can no longer get to the preferences page without having connected to a keyboard.

obra, 2020-02-14:


algernon, 2020-02-14:
Oh, shoot. I accidentally removed that part when rebasing.

algernon, 2020-02-18:
Hmm... interesting, I can't reproduce?

[image]

[image]

I can get to preferences even if I start up Chrysalis, and never connect to a keyboard, not just when I connect & disconnect.

algernon, 2020-04-29:
@obra Is this still a problem?

pm0u, 2020-05-02:
I can confirm I can also get to the preferences screen on a clean startup with no keyboard connected in the latest master

algernon, 2020-05-02:
@pm0u Thanks for testing!

Closing this then.

closed 2020-05-02 by algernon