keyboardio/Chrysalis #465: After connecting to keyboard no other menues can be selected

QEDeD, 2020-02-17:
**Describe the bug**
I can connect to the keyboard just fine, though the option to 'Disconnect' is greyed out. However, when I try to enter any other screens than the 'Connect'-screen, nothing happens when I click those items in the main menu. Clickings 'Real-time chat', 'Send feedback' and 'Exit Chrysalis' works as intended. 

**To Reproduce**
[-1581966964012.log](https://github.com/keyboardio/Chrysalis/files/4215766/-1581966964012.log)

Steps to reproduce the behavior:
1. Install and open latest Chrysalis
2. Connect Keyboard
3. Try to access any other screen/page than the 'connect'-page/start page
4. Nothing happens

**Expected behavior**
Being to use the menu to access other pages.

**Screenshots**
If applicable, add screenshots to help explain your problem. If possible, please also include a log from the "developer console". You can access it by pressing `Control + Shift + I`. When it appears, right-click inside it, and select "Save as", and attach the saved file to this report.

**Desktop (please complete the following information):**
 - OS: Windows 10 1909 (build 18363.768)
 - Chrysalis Version: 0.6.2+143

obra, 2020-02-17:
Can you try the 0.6.2 release?

QEDeD, 2020-02-17:
Thanks for the quick reply, @obra :)

Instead of getting the lastest binary via going to https://github.com/keyboardio/Chrysalis and clicking the 'Latest EXE' button which leads to a "Chrysalis-0.6.2+ 143.exe' file, I went to the releases page (https://github.com/keyboardio/Chrysalis/releases) and got an .exe from there which is just called 'Chrysalis-0.6.2.exe'. That one works fine.. which seems pretty odd to me, shouldn't both ways lead to the same file?

algernon, 2020-02-17:
> which seems pretty odd to me, shouldn't both ways lead to the same file?

"Latest" is the latest build from the master branch, while the latest release is the latest tagged release, which is usually older. Every push to master will automatically build a new installer for the three major platforms, the "latest" link points there. These are all automatic, and aren't releases per se. The release page contains installers which are... well... release builds. With changelog, and considerably more testing and polish than the automatic builds.

The automatic builds are mostly there to make it easier to test the bleeding edge, without having to compile Chrysalis.

QEDeD, 2020-02-17:
Ah, I see, thanks for making the difference clear.

algernon, 2020-02-17:
Lets keep this issue open though, because we do need to fix it!

QEDeD, 2020-02-17:
I'm happy to test things if needed.

algernon, 2020-02-18:
Perhaps this is an AppImage-only thing, but when running with `yarn run start`, I can't reproduce this. Will try in a bit with an AppImage.

algernon, 2020-02-18:
Aha! Ok, so this is the serialport bug. Ok.

algernon, 2020-04-29:
We fixed this with #467, the issue was caused by webpack and serialport not playing well together in the production build, which resulted in getting stuck, which is why the menus did not work.

closed 2020-04-29 by algernon