keyboardio/Chrysalis #474: "LockTo" and "ShiftTo" look wrong

algernon, 2020-03-05:
**Describe the bug**
The "LockTo" and "ShiftTo" top labels for layer keys feel a bit awkward.

**Expected behavior**
We should have a space between the two words.

pm0u, 2020-05-02:
I did this quickly on an atreus screen, and the text is a _little_ long on shift to and unfortunately SVGs don't support text wrapping. I'm not sure what this would look like on other keyboard models as well. It could be possible to split up the text to 2 lines, or to shrink the text ever so slightly.
[image]

obra, 2024-11-11:
This generally looks better now

closed 2024-11-11 by obra