keyboardio/Chrysalis #476: "Blocked" should be "No Action"

algernon, 2020-03-06:
**Is your feature request related to a problem? Please describe.**
We currently use "Blocked" as the label for the empty key, not very accurate, and not very friendly, either.

**Describe the solution you'd like**
Ideally, we would have "No Action" as label, on two rows, but not with a wide gap between them, like we'd have if we used the current `top` and `primary` label parts.

rudolph9, 2021-12-12:
On this note, what is the difference between "transparent" and "Blocked", it would be nice to have an explanation in the ui.

algernon, 2021-12-12:
Not quite sure how, but I agree, we should have that explained somewhere discoverable within Chrysalis.

The difference is that if a key is `Transparent` on a layer, when encountered, it will be looked up from any layer activated before the most recently activated one, while `Blocked` will simply be a no-op. So, if you have Layer 0 and Layer 1 active, and a key is `Transparent` on L1, it will be looked up from L0 instead. If it's `Blocked` on L1, then no matter what it is on L0, it will not be looked up.