keyboardio/Chrysalis #480: Firmware update requires disconnecting keyboard afterward .7

darinclarkdev, 2020-03-06:
With .7 and .71 when you do a Firmware Update the keyboard is stuck with the ESC light on afterwards. You have to disconnect/reconnect to get the keyboard functional again. I am almost certain this didnt' happen in .62

algernon, 2020-03-06:
Hm, that's interesting, because flashing wasn't changed since 0.6.2.

namecheapmohan, 2020-03-06:
In my case when updating firmware to 0.7.1 default, the app showed that the firmware update was successful, but the tab key stayed lit for a long time and the keyboard was not responding. I had to disconnect keyboard and connect again. I don't really know now what version of firmware is actually on the keyboard.

qayshp, 2020-03-07:
Same problem here since 0.7.0. Haven't had a chance to try 0.7.2 yet though.

In case it helps, my keyboard is connected using a USB-C cable directly attached to MacBook Pro running a fully patched macOS. Tried flashing both experimental and default firmware with same effect.

obra, 2020-03-07:
Algernon tracked this down today. It should be fixed in 0.7.3

algernon, 2020-03-08:
I tagged & released 0.7.3, and it Works For Me(tm). @namecheapmohan, @qayshp can either of you check 0.7.3, and verify the issue is fixed?

darinclarkdev, 2020-03-08:
yes, it does reboot after the update now. Thanks.

algernon, 2020-03-08:
Thanks for the quick testing! Closing this then.

closed 2020-03-08 by algernon