keyboardio/Chrysalis #481: Atreus flashing needs better feedback

algernon, 2020-03-06:
**Describe the bug**
Flashing the Atreus provides little feedback. When something goes wrong, it can stay in the flashing state for _minutes_ without feedback. It does not take that long. The log only contains "flashing, please wait", which is completely unhelpful.

**Expected behavior**
I'd expect flashing to work, and a reasonable timeout with proper feedback if it fails for some reason.

algernon, 2020-03-08:
This turned out to be somewhat related to #480. We got stuck on "please wait", because avrgirl-arduino was trying to verify the firmware, and was waiting for more data than what is available. That library does not have a timeout for _this_ operation, and we do not use a timeout for flashing, thus we end up waiting forever.

We probably should have a timer here, and error out after about a minute (or perhaps even after half a minute...).

obra, 2020-03-08:
That duration of a timer is fine for -verify-, but not for flash. for
flash, we should definitely have a longer timeout.

algernon, 2020-03-08:
At the moment, `avrgirl-arduino` does not allow us to separate flashing from verify, so we can either have a timeout for the whole thing (erase, program, verify, fuse-check), or work with Suz and make it possible to have separate timeouts (either by providing separate steps so we can do our own timeout, or having timeouts within avrgirl; former is likely easier).

In the short term, the first option is considerably easier... what's a reasonable timeout for the whole process?

obra, 2020-03-09:
What's the timeout Arduino uses?
ᐧ

obra, 2020-03-09:
Honestly, better state reporting is going to get us the better experience
faster.
ᐧ

algernon, 2022-05-17:
We have a stepper show now during flashing, so there's some progress feedback. I think that should cover the better feedback requirement. Closing this accordingly.

closed 2022-05-17 by algernon