keyboardio/Chrysalis #486: Cant set a modifier to be a oneshot key of (

plessbd, 2020-03-12:
**Describe the bug**
Unless I am missing it completely I cannot fine a way to use the `(` or `)` as a key.

In the .6.2 experimental firmware the shift keys would be `(` or `)`

In the .7.3 experimental version this is no longer default.

I am also unable to figure out a way to set this using the GUI as I cannot find `(` or `)` in any of the lists of keys

I am able to reenable the behavior by assigning a key to disable space cadet, but anytime i disconnect / reconnect the keyboard i have to disable space cadet

**Desktop (please complete the following information):**
 - OS: macOS
 - Chrysalis Version: .7.3

pm0u, 2020-05-03:
> I am also unable to figure out a way to set this using the GUI as I cannot find ( or ) in any of the lists of keys
I would second this and mention that on the first go, many characters were hard for me to find. As an example, I expected to find symbols such as `(` or `[` under "symbols", however `(` is under digits>9 with a `shift` modifier and `[` is under punctuation. Perhaps including these in more than one key grouping would be helpful

obra, 2020-05-03:
Including in more than one grouping is ok with me.

Alternatively, we could consider an alternate view that shows most things
where you'd find them on a normal 104 key keyboard...