keyboardio/Chrysalis #487: Improve first-run experience for key remapping

ahpook, 2020-03-12:
**Describe the bug**
In order to remap a key in Chrysalis, the user has to change default settings, either to set the default layer to a user-modifiable layer or to turn on "use only custom layers". There's no guidance for this in the UI or docs and no feedback as to why the change might not be working.

**To Reproduce**
Steps to reproduce the behavior:
1. Install and run chrysalis, select a keyboard, change a key and click "save". 
2. The UI reports the save was successful, but the change is not active because, assuming this is your first time modifying the mappings, the default layer is the factory "Layer 0", not the one you just changed.
3. Beg for help online
3a. note that Layer 3 is not just a mutable copy of 0, it has space/backspaces reversed and the palm keys did not enable layer 2 - I had to change them to ShiftTo 5

**Expected behavior**
Once you start editing, the user-defined layers ought to be made the default. 

**Desktop (please complete the following information):**
 - OS: MacOs 10.13
 - Chrysalis Version: 0.7.3+173

neuhaus, 2020-06-19:
I ran across this exact issue.

aspiers, 2020-06-27:
I'm somewhat baffled by this issue, or at least the previous comment, because I'm using 0.7.7 and I don't see this.  I'd never used Chrysalis before, and my changes immediately took effect, both with my Atreus and Model 01.  Here are the default settings:

[image]

aspiers, 2020-06-27:
Also, I didn't even know about the hardcoded layers and default layer until I read this bug and found the preferences.

neuhaus, 2020-06-27:
Ok so i don't have this issue after all. It was a mix of using Atreus with a non-US keymap

algernon, 2021-01-25:
I think we have improved this recentish-ly: we no longer support a mixture of hardcoded and eeprom-based keymaps, and Chrysalis will display a warning (with a "Fix it" button) if it encounters such a setup. As such, when one is on the editor page, any save will immediately take effect.

If the eeprom layout is empty, Chrysalis will copy the built-in keymap there first, to make the experience smoother.

I believe this issue can be closed now, but do feel free to reopen if it still needs work.

closed 2021-01-25 by algernon