keyboardio/Chrysalis #490: Windows installer does not create a start menu icon

JonathanTroyer, 2020-03-27:
**Describe the bug**
After running the installer, no start menu entry is created, making opening the app after installation impossible.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the windows installer
2. View the start menu

**Expected behavior**
A start menu entry to be created

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro 1909
 - Chrysalis Version: 0.7.4

**Additional context**
N/A

algernon, 2020-04-29:
I can't reproduce this. With both 0.7.4 and current master, on Windows 10 Pro 1909, both a desktop entry and a start menu item gets created for me.

JonathanTroyer, 2020-04-29:
RIP. The one downloaded from the README link works, but not the one downloaded directly from Releases.

closed 2020-04-29 by JonathanTroyer