keyboardio/Chrysalis #491: Add a 'debug log" link in lower right corner of screen.

obra, 2020-04-01:
I wonder if we could stick a tiny link in the lower right corner that says "Debug log" which opens a text area with the contents of the console.

algernon, 2020-04-02:
There's a few things that make this harder than one would like... first of all, the lower right corner is where the "Save" button is on the layout & colormap editor screen. We do not want to conflict with that.

The other is, I'm not sure how I can capture the console contents from the webapp. The best I can do, I think, is to listen to the `console-message` events on the webView, and display that when the debug log button is clicked. In some ways, this is likely friendlier than capturing _all_ dev console output. I will need to check if it captures everything we're interested in, though.

algernon, 2020-04-02:
Well, `console-message` is... okay-ish? As long as we log formatted messages, we're good. I can capture that stuff within the app. Now the question is, where to put the debug button and the contents?

I was thinking, we could have a key binding instead - though that's not very discoverable. Or we could provide a button on the error notification and on the keyboard select screen. On the error notification, because we're mostly interested in the logs when there are errors. On the keyboard select screen, because that's the first screen, we have space there, so we can pop the logs up there in case the error notification did not display.

algernon, 2020-04-02:
Current status on the [debug/log-viewer](https://github.com/keyboardio/Chrysalis/tree/debug/log-viewer) branch:

[image]

---

[image]

obra, 2020-04-02:
We can certainly move the button to the lower left corner to make it not
conflict with things.. And possibly make it even lighter grey

ᐧ

algernon, 2020-04-02:
Lower left might be an issue with the key group dropdown, but much less so. The dropdown will simply hide the button, so we're ok.

If we want to deemphasize it, I can remove the outline. I'm not sure we should be playing much with colors - the ones provided by material UI are in general well thought out, imo.

obra, 2020-04-02:
Yes, removing the border is great.  we can always play with the location later

algernon, 2020-04-02:
Current status:

[image]
[image]

We're a bit off on some screens though:

[image]

algernon, 2020-10-09:
I think this is obsolete with #575? The system info screen allows us to view debug logs, and even makes a bundle for easy sharing, so it's a better way to get logs to us.

obra, 2020-10-09:
yup

closed 2020-10-09 by obra