keyboardio/Chrysalis #501: Firmware update changelog display

jimt, 2020-04-18:
**Is your feature request related to a problem? Please describe.**
On the firmware update page, it would be helpful to show:
* current detected version of the firmware in the selected keyboard
* changelog (or link to changelog) so I can decide if I need/want to update

**Describe the solution you'd like**
A changelog highlighting my current version of firmware, showing significant changes between it and the version Chrysalis is proposing to update to.

**Describe alternatives you've considered**
I'm not sure how to find out my current firmware version on an Atreus.

**Additional context**
* Chrysalis-0.7.3+173.AppImage
* Ubuntu (20.04)

parke, 2021-09-21:
There is some related discussion in the final paragraphs of this comment:

https://github.com/keyboardio/Chrysalis/issues/702#issuecomment-891363761

I, too, would like Chrysalis to proactively auto-detect and report if the keyboard is running an out-of-date version of the Chrysalis-provided firmware.

If the keyboard firmware(s) is (are) out of date, then an up-to-date version of Chrysalis can install configurations that **will not work correctly**.  This can be frustrating for the end user to diagnose.  I definitely consider it to be a bug.

algernon, 2022-05-27:
We have a Changelog screen now, which shows all user-visible changes in Chrysalis itself, and also notes when firmware is updated. In general, we try to keep firmware backwards compatible, so upgrading to the latest should be considered safe. Staying on an older one should also be considered safe, because we also try our best to not break existing setups.

In rare cases we do, Chrysalis will offer a migration path.

As such, I do not think that detecting the firmware version is of any use, especially not when Chrysalis works with custom firmware too, that have no relation whatsoever to the firmware it ships with (apart from running Kaleidoscope on supported hardware, and having a few basic plugins enabled, of course). Firmware version detection would only be useful for those who run the stock firmware. And in those cases, upgrading or staying on whatever one has are both safe options.

I don't think it is necessary to nag people to upgrade their firmware. Therefore, I think that having a few notes about firmware changes in the Chrysalis changelog (which is now available via the application itself) is sufficient.

On top of that:

> If the keyboard firmware(s) is (are) out of date, then an up-to-date version of Chrysalis can install configurations that **will not work correctly**. This can be frustrating for the end user to diagnose. I definitely consider it to be a bug.

This is a valid concern, which is why #837 exists. When merged, that'll stop Chrysalis - in its default configuration - from installing configurations that do not work with the firmware on the device. At least as long as the firmware isn't terribly old, and supports the `plugins` command.

closed 2022-05-27 by algernon