keyboardio/Chrysalis #514: Support more XF86 Keycodes

pm0u, 2020-05-03:
**Is your feature request related to a problem? Please describe.**
I would like to be able to control the brightness of my laptop from my external keyboard. This would be accomplished by sending the XF86MonBrightnessUp/Down keycodes. There are many XF86 keycodes that others may find useful as well.

**Describe the solution you'd like**
Additional support for these keycodes. It looks like _some_ are supported, such as a few media functions. There is a list [here](http://wiki.linuxquestions.org/wiki/XF86_keyboard_symbols).

I think having these all as options, or the ability to input the hex codes found [here](https://cgit.freedesktop.org/xorg/proto/x11proto/tree/XF86keysym.h) would work as well. It does not seem like the "Unknown Keycodes" option is intended for this type of key symbol.

obra, 2020-05-03:
Do we know what USB HID key codes these send?

pm0u, 2020-05-03:
> Do we know what USB HID key codes these send?
> […](#)

That's a good question. I do not, hardware is not my area of expertise. If something like `xev` output or if there's another Linux utility to figure this out, I'd be happy to provide that.

obra, 2020-05-03:
xev is absolutely a fine start, though it may or may not give us sufficient data.

pm0u, 2020-05-03:
This is the xev output for up then down

```
MappingNotify event, serial 40, synthetic NO, window 0x0,
    request MappingKeyboard, first_keycode 8, count 248

FocusOut event, serial 40, synthetic NO, window 0x4c00001,
    mode NotifyGrab, detail NotifyAncestor

FocusOut event, serial 41, synthetic NO, window 0x4c00001,
    mode NotifyUngrab, detail NotifyPointer

FocusIn event, serial 41, synthetic NO, window 0x4c00001,
    mode NotifyUngrab, detail NotifyAncestor

KeymapNotify event, serial 41, synthetic NO, window 0x0,
    keys:  0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   

FocusOut event, serial 41, synthetic NO, window 0x4c00001,
    mode NotifyGrab, detail NotifyAncestor

FocusOut event, serial 41, synthetic NO, window 0x4c00001,
    mode NotifyUngrab, detail NotifyPointer

FocusIn event, serial 41, synthetic NO, window 0x4c00001,
    mode NotifyUngrab, detail NotifyAncestor

KeymapNotify event, serial 41, synthetic NO, window 0x0,
    keys:  2   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0  
```

pm0u, 2020-05-03:
It looks like those hex codes line up with the doc i shared above, is the `keycode` value what we are looking for?

obra, 2020-05-03:
Hm. That's not the output I'd expect from xev.

Can you try logging:

Tap the letter "a"

Tap a -single- XF86 special key.

Tap the letter "b"

?

pm0u, 2020-05-03:
here ya go (this is with hotkey daemon disabled) and pressed monbrightnessup:
```
KeyPress event, serial 34, synthetic NO, window 0x600001,
    root 0x1a0, subw 0x0, time 11122084, (-28,604), root:(1256,624),
    state 0x0, keycode 38 (keysym 0x61, a), same_screen YES,
    XLookupString gives 1 bytes: (61) "a"
    XmbLookupString gives 1 bytes: (61) "a"
    XFilterEvent returns: False

KeyRelease event, serial 34, synthetic NO, window 0x600001,
    root 0x1a0, subw 0x0, time 11122162, (-28,604), root:(1256,624),
    state 0x0, keycode 38 (keysym 0x61, a), same_screen YES,
    XLookupString gives 1 bytes: (61) "a"
    XFilterEvent returns: False

MappingNotify event, serial 34, synthetic NO, window 0x0,
    request MappingKeyboard, first_keycode 8, count 248

KeyPress event, serial 34, synthetic NO, window 0x600001,
    root 0x1a0, subw 0x0, time 11124454, (-28,604), root:(1256,624),
    state 0x0, keycode 233 (keysym 0x1008ff02, XF86MonBrightnessUp), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 34, synthetic NO, window 0x600001,
    root 0x1a0, subw 0x0, time 11124454, (-28,604), root:(1256,624),
    state 0x0, keycode 233 (keysym 0x1008ff02, XF86MonBrightnessUp), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

MappingNotify event, serial 35, synthetic NO, window 0x0,
    request MappingKeyboard, first_keycode 8, count 248

KeyPress event, serial 35, synthetic NO, window 0x600001,
    root 0x1a0, subw 0x0, time 11126327, (-28,604), root:(1256,624),
    state 0x0, keycode 56 (keysym 0x62, b), same_screen YES,
    XLookupString gives 1 bytes: (62) "b"
    XmbLookupString gives 1 bytes: (62) "b"
    XFilterEvent returns: False

KeyRelease event, serial 36, synthetic NO, window 0x600001,
    root 0x1a0, subw 0x0, time 11126402, (-28,604), root:(1256,624),
    state 0x0, keycode 56 (keysym 0x62, b), same_screen YES,
    XLookupString gives 1 bytes: (62) "b"
    XFilterEvent returns: False
```

pm0u, 2020-05-03:
`evtest` outputs this:
```
    Event code 224 (KEY_BRIGHTNESSDOWN)
    Event code 225 (KEY_BRIGHTNESSUP)
```
however, when i press these keys nothing happens (it doesn't register a key event)

obra, 2020-05-03:
For context, we're trying to find the mappings into
https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf for those
XF86 keycodes.

It's entirely possible that there -isn't- a mapping, though.

pm0u, 2020-09-30:
Hey @obra sorry i dropped this -- if it's helpful, I have a Keychron K1 that seems to successfully control brightness via USB... is there something I could capture to help figure this out?

obra, 2020-10-04:
This is the full list of new HID codes available to us. Some of them need to be brought into Chrysalis.

The doc describing them is here: https://usb.org/document-library/hid-usage-tables-12

HID_INDEX_TRIGGER 0x94
HID_PALM_TRIGGER 0x95
HID_THUMBSTICK 0x96
HID_SYSTEM_FUNCTION_SHIFT 0x97
HID_SYSTEM_FUNCTION_SHIFT_LOCK 0x98
HID_SYSTEM_FUNCTION_SHIFT_LOCK_INDICATOR 0x99
HID_DISMISS_NOTIFICATION 0x9A
HID_DO_NOT_DISTURB 0x9B
HID_CONSUMER_CONSUMER_CONTROL_CA 0x01
HID_CONSUMER_PICTURE_IN_PICTURE_TOGGLE 0x67
HID_CONSUMER_PICTURE_IN_PICTURE_SWAP 0x68
HID_CONSUMER_RED_MENU_BUTTON 0x69
HID_CONSUMER_GREEN_MENU_BUTTON 0x6A
HID_CONSUMER_BLUE_MENU_BUTTON 0x6B
HID_CONSUMER_YELLOW_MENU_BUTTON 0x6C
HID_CONSUMER_ASPECT 0x6D
HID_CONSUMER_3D_MODE_SELECT 0x6E
HID_CONSUMER_DISPLAY_BRIGHTNESS_INCREMENT 0x6F
HID_CONSUMER_DISPLAY_BRIGHTNESS_DECREMENT 0x70
HID_CONSUMER_DISPLAY_BRIGHTNESS 0x71
HID_CONSUMER_DISPLAY_BACKLIGHT_TOGGLE 0x72
HID_CONSUMER_DISPLAY_SET_BRIGHTNESS_TO_MINIMUM 0x73
HID_CONSUMER_DISPLAY_SET_BRIGHTNESS_TO_MAXIMUM 0x74
HID_CONSUMER_DISPLAY_SET_AUTO_BRIGHTNESS 0x75
HID_CONSUMER_CAMERA_ACCESS_ENABLED 0x76
HID_CONSUMER_CAMERA_ACCESS_DISABLED 0x77
HID_CONSUMER_CAMERA_ACCESS_TOGGLE 0x78
HID_CONSUMER_KEYBOARD_BRIGHTNESS_INCREMENT 0x79
HID_CONSUMER_KEYBOARD_BRIGHTNESS_DECREMENT 0x7A
HID_CONSUMER_KEYBOARD_BACKLIGHT_SET_LEVEL 0x7B
HID_CONSUMER_KEYBOARD_BACKLIGHT_OOC 0x7C
HID_CONSUMER_KEYBOARD_BACKLIGHT_SET_MINIMUM 0x7D
HID_CONSUMER_KEYBOARD_BACKLIGHT_SET_MAXIMUM 0x7E
HID_CONSUMER_KEYBOARD_BACKLIGHT_AUTO 0x7F
HID_CONSUMER_VOICE_COMMAND 0xCF
HID_CONSUMER_INVOKE_CAPTURE_INTERFACE 0xD0
HID_CONSUMER_START_OR_STOP_GAME_RECORDING 0xD1
HID_CONSUMER_HISTORICAL_GAME_CAPTURE 0xD2
HID_CONSUMER_CAPTURE_GAME_SCREENSHOT 0xD3
HID_CONSUMER_SHOW_OR_HIDE_RECORDING_INDICATOR 0xD4
HID_CONSUMER_START_OR_STOP_MICROPHONE_CAPTURE 0xD5
HID_CONSUMER_START_OR_STOP_CAMERA_CAPTURE 0xD6
HID_CONSUMER_START_OR_STOP_GAME_BROADCAST 0xD7
HID_CONSUMER_AL_MESSAGE_STATUS 0x1C8
HID_CONSUMER_AL_CONTACT_SYNC 0x1C9
HID_CONSUMER_AL_NAVIGATION 0x1CA
HID_CONSUMER_AL_CONTEXT_AWARE_DESKTOP_ASSISTANT 0x1CB
HID_CONSUMER_AC_NEXT_KEYBOARD_LAYOUT_SELECT 0x29D
HID_CONSUMER_AC_NAVIGATION_GUIDANCE 0x29E
HID_CONSUMER_AC_DESKTOP_SHOW_ALL_WINDOWS 0x29F
HID_CONSUMER_AC_SOFT_KEY_LEFT 0x2A0
HID_CONSUMER_AC_SOFT_KEY_RIGHT 0x2A1
HID_CONSUMER_AC_DESKTOP_SHOW_ALL_APPLICATIONS 0x2A2
HID_CONSUMER_AC_IDLE_KEEP_ALIVE 0x2B0
HID_CONSUMER_EXTENDED_KEYBOARD_ATTRIBUTES_COLLECTION 0x2C0
HID_CONSUMER_KEYBOARD_FORM_FACTOR 0x2C1
HID_CONSUMER_KEYBOARD_KEY_TYPE 0x2C2
HID_CONSUMER_KEYBOARD_PHYSICAL_LAYOUT 0x2C3
HID_CONSUMER_VENDOR_SPECIFIC_KEYBOARD_PHYSICAL_LAYOUT 0x2C4
HID_CONSUMER_KEYBOARD_IETF_LANGUAGE_TAG_INDEX 0x2C5
HID_CONSUMER_IMPLEMENTED_KEYBOARD_INPUT_ASSIST_CONTROLS 0x2C6
HID_CONSUMER_KEYBOARD_INPUT_ASSIST_PREVIOUS 0x2C7
HID_CONSUMER_KEYBOARD_INPUT_ASSIST_NEXT 0x2C8
HID_CONSUMER_KEYBOARD_INPUT_ASSIST_PREVIOUS_GROUP 0x2C9
HID_CONSUMER_KEYBOARD_INPUT_ASSIST_NEXT_GROUP 0x2CA
HID_CONSUMER_KEYBOARD_INPUT_ASSIST_ACCEPT 0x2CB
HID_CONSUMER_KEYBOARD_INPUT_ASSIST_CANCEL 0x2CC
HID_CONSUMER_PRIVACY_SCREEN_TOGGLE 0x2D0
HID_CONSUMER_PRIVACY_SCREEN_LEVEL_DECREMENT 0x2D1
HID_CONSUMER_PRIVACY_SCREEN_LEVEL_INCREMENT 0x2D2
HID_CONSUMER_PRIVACY_SCREEN_LEVEL_MINIMUM 0x2D3
HID_CONSUMER_PRIVACY_SCREEN_LEVEL_MAXIMUM 0x2D4

Rom1deTroyes, 2020-10-05:
I can try to do this (will be a start !).
I look at the code this morning, and will ask questions on Discord if I don't find my way :-)

gedankenexperimenter, 2020-10-05:
You probably want `HID_CONSUMER_DISPLAY_BRIGHTNESS_INCREMENT` and `HID_CONSUMER_DISPLAY_BRIGHTNESS_DECREMENT`, which correspond to the `Key` values `Consumer_DisplayBrightnessIncrement` and `Consumer_DisplayBrightnessDecrement`.

Rom1deTroyes, 2020-10-06:
I made a first try in #569 with `HID_CONSUMER_DISPLAY_BRIGHTNESS_INCREMENT` and `HID_CONSUMER_DISPLAY_BRIGHTNESS_DECREMENT` (leaving them in the `Media` group for now).

I don't have hardware to test this, but the keys are reported to be the good ones under my two GNU/Linux boxes :thinking:

algernon, 2021-01-25:
I think this can be closed: #569 should have addressed this, and even after the keycode database rework, brightness up/down still appears to be selectable.

closed 2021-01-25 by algernon