keyboardio/Chrysalis #518: 0.75 on Windows -  serial port not redetected after flashing

obra, 2020-05-18:
0.75 on Win 10.

Flashed an Atreus, including wiping EEPROM.

After flashing, I get to the "Can't talk the keyboard over Focus. You need new firmware"
screen.

The problem appears to be that after flashing, the keyboard is on yet another new COM port.

Going to the "Connect to a different keyboard" screen and forcing redetection, it finds the new com port and everything is fine:

[image]

algernon, 2020-05-19:
There are a number of things in play here, which contribute to ending up in such a confusing situation. Lets look at the logs first: 

- In the `COM5` case, the `serialNumber` is `atreus`, and `pnpId` lists the product id as `2302` (which is bootloader mode).
- In the `COM6` case, the `serialNumber` is `CatreusE` and `pnpId` lists the product id as `2303` (normal mode).
- When trying to reconnect, we first get `access denied` 3 times, then `file not found` until re-detection.

What this means, is that after flashing, Windows was slow to reattach the device (which is kind of expected, Windows is incredibly slow at attaching new devices, I found), so you ended up trying to connect to a keyboard that was booting the sketch, which didn't work, because the firmware was busy booting, and then Windows reattached the device onto a different port.

We have two issues that we can fix on the Chrysalis side:

- We should signal whether a detected device is in bootloader mode or not, if we can figure that out.
- We should look into why automatic re-detection didn't kick in, while hitting the re-detect button was. The detection code is the same, so I _think_ we fail to listen to new USB events on Windows.

The latter can be tested reasonably easily: if you unplug a keyboard, does it disappear from the list after a short while (after Windows notices it is unpluged, that is)? If you plug it back in, does it get discovered once Windows finishes attaching it? Or do you need to hit the re-detect button for both?

A third thing we _could_ try doing, but I think we shouldn't, is port re-detection during connect. The dropdown shows the port the device is connected to, and we're trying to connect there. If we re-detect the port in the process, we may end up connecting to a different device than intended. We were asked to connect to a specific port, so we should do that, and not second-guess. We've done this to support multiple keyboards of the same type being connected, and the user being able to choose which one to connect to (even though it is non-trivial to figure out which one's which, unless you already know what port they're connected to). Re-detecting the port would effectively result in "connect to a possibly random keyboard of this type", which would be a worse user experience in my opinion.

obra, 2020-05-20:
>We should signal whether a detected device is in bootloader mode or not, if we can figure that out.

At least for Keyboardio boards, that's going to be obvious from usb device id.

> A third thing we could try doing, but I think we shouldn't, is port re-detection during connect. 

After flashing, I think we need to treat the keyboard as having been fully disconnected, so we -should- be throwing the user back to the "pick a keyboard" screen.

> The dropdown shows the port the device is connected to, and we're trying to connect there. If we re-detect the port in the process, we may end up connecting to a different device than intended.

From a "technically correct" viewpoint, I agree with you. From a "works for customers" viewpoint, I disagree. If we can't connect and think we should be able to, redetection is going to make users happier and less frustrated. Showing an informational notification that we connected to a different port is fine, though.

Once we're on fancier MCUs, they'll actually have unique hardware serial numbers.

algernon, 2020-05-20:
> After flashing, I think we need to treat the keyboard as having been fully disconnected, so we -should- be throwing the user back to the "pick a keyboard" screen.

That is what _should_ be happening. If it doesn't, that is a bug.

> From a "works for customers" viewpoint, I disagree. If we can't connect and think we should be able to, redetection is going to make users happier and less frustrated. Showing an informational notification that we connected to a different port is fine, though.

That is a convincing argument. How about a setting in advanced settings where we can turn this _off_ (and consequently, default to redetection during connect)? Making it configurable is minimal overhead, and would be useful for development where I do want to keep the port stable.

obra, 2020-05-20:
Having an advanced setting to give a developer better development tools is
fine.

algernon, 2022-05-27:
Since we're pretty reliable throwing ourselves back to the keyboard selection screen after flashing (successful or not), I think this issue can be safely closed.

There is the edge case of the dropdown showing one port, and the keyboard moving to another during connect, but I feel that's a very, very rare edge case. If we need to handle that, I'd prefer a new issue about it, and about that only. (And it wouldn't be windows-only, same thing can happen on any platform)

closed 2022-05-27 by algernon