keyboardio/Chrysalis #519: Deleted layers 1 & 2 on first attempt to save

a1tsal, 2020-05-25:
I'm running on MacOS 10.14.6. Installed Chrysalis 0.7.5+221 today. Keyboardio 01. (Don't know what firmware it is running; can't see a way to get it in the Chrysalis UI.) 

First thing I did after installing was to change some key assignments in the UI. Then I pressed the Save Changes button. It turned green, but the layout editor went pale gray and blank.

Poking around a bit, I figured out that Layers 0 and 1 had been deleted. (I did not push the Clear Layer button.) The layers menu starts at 3.

The keyboard itself seems to be unaffected (no layout change but is working fine).

Command-shift-I has no effect.

baconisthecure, 2020-06-05:
I'm experiencing same.

obra, 2020-06-05:
Do you also see this on 0.75? 

If you go into Preferences and turn on display of hardcoded layers and turn off “use custom layers only” and hit save, are your missing layers visible again?

a1tsal, 2020-06-06:
Installed 0.75. Went to Preferences, turned on hardcoded layer display ("custom only" was already off) and saved.

The missing layers (0, 1, 2) now display, with a lock icon in the dropdown menu. All three layers are displayed grayed-out and cannot be modified. Layers 3+ have no lock icon and are not grayed-out.

I modified a key on Layer 7 and hit the save button. The save button showed a rotating green progress indicator for several minutes. I hit the X-to-close button on the layout editor and it popped up a modal "Cancel pending changes?" I selected OK but the save button continued rotating and the layout editor did not close.

Is there a way to restore the state to the starting point, rather than trying to debug from this broken configuration?

a1tsal, 2020-06-06:
Also, is there a known-working version available for download?

obra, 2020-06-06:
> 
> 
> Also, is there a known-working version available for download? 
Other than the two folks on this ticket, my understanding was that 0.75 worked.

I don’t believe the issue you’re seeing is with the keyboard side of things, but with Chrysalis. 

To reset the keyboard, you can go into Preferences -> Advanced and click the “Reset EEPROM” button. 

Please do report back on whether that works for you.

a1tsal, 2020-06-06:
Thanks! I reset the EEPROM.

Going back to the layout editor, that had shifted the layer numbers all down by three. So now instead of layers 0 to 7, there were layers -3 to 4. The negative-numbered layers were locked.

I modified layer 0 and did a save. That shifted the layer numbers back again (so I have 0 to 7) and layers 0-2 are locked.

The keyboard was apparently unaffected (the key changes I made did not take effect).

a1tsal, 2020-06-06:
I thought I'd try updating the firmware. (I did not do this before because I didn't see it in the instructions... if it's there and I missed it, my bad!)

That said it succeeded.... and now I have a completely different problem. I'll follow up either here or in a new issue, once I figure it out some more.

a1tsal, 2020-06-06:
OK, so the visible effect of the firmware update was that I got layers 0-4 (only 5 layers vs 7 previously). The UI was showing "Colormap Editor" and all the keys were solid black and unlabeled. I couldn't see any way to get to the layout editor.

I quit the program and restarted.

So then it showed layers 3-7. No layers 0-2.

However, the UI shows "Layout & Colormap Editor" (so that's a step in the right direction I guess). 

Layer 3 shows the default key symbols. I modified some keys on that layer and saved.

This had no effect on the actual keyboard (i.e. the keys I changed in Chrysalis continued to have their default meanings.)

a1tsal, 2020-06-06:
Just to be thorough, I restarted the program, redid the firmware update, and redid the EEPROM reset. UI as before (layers 3-7. Modified a key, saved; it said it succeeded, but did not take effect in physical reality.

obra, 2020-06-06:
Ok. I’ve managed to reproduce the issue and I think that I understand what’s going on.

In the Preferences pane, we have a few related settings.

“Show hardcoded layers”
“Use custom layers only”
“Default layer”

The idea behind these settings is that you might have some key layouts defined in the firmware that was flashed to the keyboard. These are the “default” layers.

“Use custom layers only” should be mandatory if “Show hardcoded layers” is not enabled. 

When you see layers 0,1,2 vanished, you’ve got “show hardcoded layers” off and -also- have “use custom layers only off"

“Show hardcoded layers” appears to change whether the hardcoded layers get shown with negative numbers or as 0,1,2.

The original reason to have this configurable involved making it easier to copy the hardcoded layers to become editable layers at first run.

This is all horribly confusing, though I’m not sure why you’re both getting non-default settings for these two selectors.

If you set “use custom layers only” to on and “show hardcoded layers” to off and hit “Save changes”, do things start to behave as expected?

a1tsal, 2020-06-06:
Hooray! Yes, with that setting, it seems to be fully functioning as expected.

Thank you very much!

baconisthecure, 2020-06-08:
Great.  It's working.    Thank you.

algernon, 2022-05-17:
This was an issue that wasn't really a bug, but Chrysalis offering a selection of options that ended up being incredibly confusing. We no longer do that, so I'm closing this.

closed 2022-05-17 by algernon