keyboardio/Chrysalis #524: Linux arm64 support

jwillikers, 2020-06-08:
**Is your feature request related to a problem? Please describe.**
I'd like to update my keyboard firmware using Chrysalis on my Pinebook Pro, which runs Manjaro on arm64 processors.

**Describe the solution you'd like**
I would like to be able to download an AppImage for arm64 and run it on the Pinebook Pro.

**Describe alternatives you've considered**
I could use another computer, use the Arduino IDE, emulate x86_64, or attempt to compile from source. However, none of these alternatives are ideal or convenient for an end user.

skyfaller, 2020-07-10:
Have you tried compiling from source by installing https://aur.archlinux.org/packages/chrysalis-git/ from AUR via e.g. [yay](https://github.com/Jguer/yay)? Just curious if it works.

Clearly it would be more convenient to have a binary for ARM64, but when AUR works, compiling from source just requires some waiting. I've had to compile a lot of more obscure software for my PBP. It's when the AUR package doesn't compile successfully, and I have no idea why or how to fix it, that I start to get frustrated and wish there were an official binary.

jwillikers, 2020-07-11:
@skyfaller That's a great idea, thanks for the suggestion. 👍
Unfortunately, no dice. 🎲🎲
I changed the architecture to `any` but the script ultimately failed.
Pulling down the latest source and using `yarn build:linux` creates an AppImage which works like a charm.

algernon, 2020-10-09:
Things build fine on Travis, all I have to do now is fix the uploads, so that the Linux x64 and amd64 builds don't overwrite each other.

commit f11d72ee (2020-10-09): Add support for Linux/arm64 builds

Travis has a Linux/arm64 environment, Chrysalis builds there fine, and works
well on arm64, too. As such, adjust our Travis config to build it, and the
artifact upload/download tools to support multiple architectures for Linux.

Fixes #524.
PR #576 (2020-10-09): Add support for Linux/arm64 builds
Travis has a Linux/arm64 environment, Chrysalis builds there fine, and works well on arm64, too. As such, adjust our Travis config to build it, and the artifact upload/download tools to support multiple architectures for Linux.

Fixes #524.

For build logs, see [here](https://travis-ci.org/github/keyboardio/Chrysalis/builds/734290991).
closed 2020-10-09 by obra