keyboardio/Chrysalis #525: MouseKeys settings customization on UI

phongnh, 2020-06-11:
Could you please add UI for customizing MouseKeys settings like `speed`, `speedDelay` and `accelSpeed` and `accelDelay`?

Default mouse settings is quite slow.

rudolph9, 2021-12-12:
[Here is the relevant doc in the Kaleidoscope repo]
(https://github.com/keyboardio/Kaleidoscope/blob/ea45b7807164c18b0df28c74c045672d12b50f18/plugins/Kaleidoscope-MouseKeys/README.md)

rudolph9, 2021-12-12:
A work around is the build custom firmware with the acceleration increased https://github.com/keyboardio/Chrysalis-Firmware-Bundle

```
diff --git a/Keyboardio/Atreus/default/Atreus/Atreus.ino b/Keyboardio/Atreus/default/Atreus/Atreus.ino
index f0f0029..7b234a4 100644
--- a/Keyboardio/Atreus/default/Atreus/Atreus.ino
+++ b/Keyboardio/Atreus/default/Atreus/Atreus.ino
@@ -139,6 +139,7 @@ void setup() {
   Kaleidoscope.setup();
   SpaceCadet.disable();
   EEPROMKeymap.setup(10);
+  MouseKeys.accelSpeed = 6;
 }

 void loop() {
```

phongnh, 2021-12-13:
Thanks @rudolph9

rudolph9, 2021-12-13:
Not to derail the purpose of this issue, but another feature I wish existed was a speedup modifier for MouseKeys. Basically a key that when pressed will make the mouse keys faster, when you stop pressing it it's slow again.

gedankenexperimenter, 2022-02-23:
@algernon — I'm looking into making some substantial modifications to MouseKeys right now, and it would be great to enable it to be configured via Chrysalis (maybe via a MouseKeysConfig plugin?).  The configuration parameters would be different than the current ones (and hopefully more useful).

algernon, 2022-02-23:
Enabling it via `MouseKeysConfig` would be great, yep. Since the current settings aren't very intuitive to begin with, exposing more useful ones would be even better.

gedankenexperimenter, 2022-02-23:
What I've got in mind is a "(maximum) speed" value and a "time to reach (maximum) speed".  The latter would be entered in milliseconds, and the former would be a somewhat abstract number.

I'm also thinking of adding some modifier keys that change the mouse speed on the fly (i.e. "hold down this key to double speed; hold down this other key to get half speed").  Those could just be done as a Macro, of course.

I'm guessing `MouseKeysConfig` would use `onSetup()` and `onFocusEvent()` to interact with EEPROM/Chrysalis?  Now that I'm thinking about it, this is something that might be best configured on the fly with an `onKeyEvent()` handler that can adjust the parameters quickly (in addition to reading & setting the value via Chrysalis, not in place of it).

PR #1143 (2022-10-13): Add Mouse Key configuration support to Preferences
[image]

Fixes #525 and #1174.

Also adds the new mousekeys commands to the list of commands to back up & restore during flash.
closed 2022-10-13 by obra