keyboardio/Chrysalis #531: Blank screen after flash clean firmware on Atreus (0.75)

obra, 2020-06-23:
I recognize that 0.75 is not current, but wanted to capture an issue I'd seen reported but never experienced myself.

After flashing an Atreus on macOS on 0.75, including wiping the EEPROM, I clicked "connect, saw the blank atreus keyboard layout indicating loading, and then the screen when white.

This is what was in the dev tools:

[image]

algernon, 2020-10-09:
I _think_ this happened due to partial loading. With the new loading screen, this shouldn't happen. I'm inclined to close this, and reopen if someone manages to reproduce the problem again.

obra, 2020-10-09:
ok

closed 2020-10-09 by obra