keyboardio/Chrysalis #533: way to identify firmware version currently on device

qayshp, 2020-06-25:
**Is your feature request related to a problem? Please describe.**
Not so much a problem as a small annoyance.
I have no way to know which firmware is currently running on my Keyboardio (flashed from Chrysalis). For example the version, `0.7.7` vs `0.7.6`, or `default` vs `experimental`.

**Describe the solution you'd like**
Some kind of indication of what was last flashed to the device.
For example:
-  `0.7.7 experimental`
-  `0.7.6 default`

**Describe alternatives you've considered**
Writing down or remembering what I did last.

**Additional context**
I know there will be complications with custom firmware bundles.

algernon, 2020-10-09:
The easiest way to do this is to embed the version in the firmware, and query that from Chrysalis. I have a [plugin](https://git.madhouse-project.org/algernon/Kaleidoscope-Focus-Version) that makes this slightly easier, but it's a bit tricky to use in official firmware.

algernon, 2022-06-24:
We have a FirmwareVersion plugin now, which will be enabled in the firmwares Chrysalis ships with. Once the updated firmware files are merged, I'll rework the FirmwareUpdate screen a little to show the currently running firmware version too (if available).

algernon, 2022-06-26:
A little progress update:

[image]

PR #966 (2022-07-01): Firmware upgrade/redesign
There were multiple complaints against the firmware update screen, and this redesign addresses most of them. The goal was to simplify the screen, reduce the amount of unnecessary information displayed at a time, while also showing useful information we did not show before.

## The main Firmware Update screen

[image]

With this redesign, the update instructions are no longer shown directly on the firmware update screen. Instead, when one selects the "Upload" or "Factory reset" actions, the popup dialog will show the instructions instead. Previously, clicking on "Uploa [... cut]
closed 2022-07-01 by obra