keyboardio/Chrysalis #541: FR: add left/right Meta to Modifiers & Locks

aspiers, 2020-06-27:
There is a subtle but important difference between a Meta key and an Alt key.  Currently Chrysalis allows binding of keys to left/right Alt (and Super), but not Meta.  It would be great to add these, and I assume pretty easy too.

obra, 2020-06-27:
I believe the canonical USB HID name for that key is "GUI", at least per
the notes in https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf
(Page
59)

I believe those keys are available, but clearly there's a usability issue
with the naming. We've briefly talked about platform-specific naming for
those keys based on the running environment.

aspiers, 2020-06-29:
Ohhhh OK, *that's* what "GUI" was for.  Yeah, definitely not intuitive ;-)  Editing title accordingly.

aspiers, 2020-12-31:
I've just tried this and AFAICS the GUI modifier is actually acting as Super not Meta.

aspiers, 2020-12-31:
I just looked at page 59 of that spec which you referenced and GUI refers to the Windows key which often equates to Super in the X11 standard.  In contrast, Meta in the X11 standard is closer (but not identical) to the modern day Alt key.

obra, 2020-12-31:
"Meta" isn't a key that's part of the USB spec. On a regular keyboard, what
are the labels on left and right Meta?

aspiers, 2020-12-31:
Meta isn't on regular keyboards these days.  But that hasn't stopped Linux and other UNIX-like systems from continuing to use them, typically by interpreting one or more Alt keys as Meta, and treating them as the `Mod1` [modifier](https://wiki.archlinux.org/index.php/X_keyboard_extension#Meta,_Super_and_Hyper).

See for instance:

- https://en.wikipedia.org/wiki/Meta_key
- https://en.wikipedia.org/wiki/Super_key_(keyboard_button)#Linux_and_BSD
- https://manpages.ubuntu.com/manpages/bionic/man1/xkeycaps.1.html#loser%20vendors although this page probably hasn't been updated in something like 15-20 years by this point (e.g. as witnessed by the fact that S.u.S.E. was renamed to SUSE around 2004 or so).

This tradition has even spilled over into Mac OS, e.g. https://www.shell-tips.com/mac/meta-key/

In practice I find that some software on Linux responds to the `Alt_{L,R}` [keysyms](https://wiki.archlinux.org/index.php/X_keyboard_extension#Keysyms_and_states), some responds to `Meta_{L,R}`, and some responds to both.

obra, 2020-12-31:
Right. Kaleidoscope and Chrysalis deal in things that can be sent over USB.
We don't have control over whether your computer's keyboard input stack is
then doing something to change them into something else.  Put another way -
we control the input into XKB, but not XKB's configuration.

aspiers, 2020-12-31:
D'oh, of course.  XKB allows a raw scancode to be interpreted as a keycode representing (roughly) the label on the corresponding physical keycap, and in fact on my system I see:

```
$ grep -ir meta /usr/share/X11/xkb/keycodes/
/usr/share/X11/xkb/keycodes/xfree86:    // <I1C> =   156;       <META>
/usr/share/X11/xkb/keycodes/xfree86:    <META> =   156; // <I1C>
/usr/share/X11/xkb/keycodes/evdev:      <META> =   205;
```

so I suppose one approach would be to make the firmware send 156 or 205, but if it's not in the USB HID spec then I guess that doesn't make sense.  Thanks for helping me see the light ;-)

closed 2020-12-31 by aspiers