keyboardio/Chrysalis #542: Cannot run Chrysalis release version 0.7.8 AppImage on Manjaro

tofraley, 2020-07-03:
**Describe the bug**
I get the errors below when trying to run the version Chrysalis-0.7.8.AppImage.
I am able to run version 0.7.7, as well as other AppImage apps.

**To Reproduce**
1. Download appimage from release page.
2. Change permissions to allow execution. 
`chmod +x Chrysalis-0.7.8.AppImage`
3. Run.
`./Chrysalis-0.7.8.AppImage`

**Expected behavior**
App runs and gui opens.

**Screenshots**
```
> ./Chrysalis-0.7.8.AppImage
Something went wrong trying to read the squashfs image.

Cannot mount AppImage, please check your FUSE setup.
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information
```

**Desktop:**
Manjaro Linux 20.0.3
Architecture: x86_64
CPU op-mode(s): 32-bit, 64-bit

**Chrysalis Version**
0.7.8

obra, 2020-07-03:
Thanks for the report. We're on it.

algernon, 2020-07-03:
This should be fixed with the 0.7.9 release.

closed 2020-07-03 by algernon